z88dk/z88dk #2189: Classic library fscanf and sscanf both broken for zsdcc

imneme, 2023-02-05:
The enclosed trivial test program works fine with sccz80, but fails to work on zsdcc.

When compiled with sccz80, we see the following (expected) results (when `input.txt` contains the word `Hello`):
```
unix% zcc +cpm hello-f.c -o hello-f.com
unix% tnylpo hello-f
Testing sscanf...
numread = 2, name = World, i = 17, offset = 8
File opened, fp = 1f92
Hello, World!
```

But when compiled with zsdcc, we get:
```
unix% zcc +cpm -compiler=sdcc hello-f.c -o hello-f.com
unix% tnylpo hello-f
Testing sscanf...
numread = 2, name = World, i = 30326, offset = 30326
File opened, fp = 2233
tnylpo: HALT instruction executed
```

This is true from the current nightly (v20577-a5bb211b71-20230205) back to release versions. 2.2 behaves the same, and 2.1 fails to even read `name` in sscanf.

Note, I tried seeing if 
```
#pragma scanf "%2d %u %x %c %s %B %o %ld %lu %lx %lc %ls %lB %lo %i %li %n"
```
made any difference but it appears not.

```c++
#include <stdio.h>
#include <assert.h>

int main(void) {
    char name[100];
    FILE* fp;
    int i = 0;
    int offset = 0;
    int numread = 0;
    printf("Testing sscanf...\n");
    numread = sscanf("World 17", "%s %d %n", name, &i, &offset);
    printf("numread = %d, name = %s, i = %d, offset = %d\n",
	   numread, name, i, offset);
    fp = fopen("input.txt", "r");
    assert(fp != 0);
    printf("File opened, fp = %p\n", fp);
    fscanf(fp, "%s", name);
    printf("Hello, %s!\n", name);

    return 0;
}
```

suborb, 2023-02-06:
Thanks for the report + test case.

Should be fixed now - looks like I got bored and switched off whilst writing the wrappers.

closed 2023-02-06 by suborb

imneme, 2023-02-07:
Thanks for the quick response.

I tried the latest nightly, and I see this:
```
Testing sscanf...
numread = 2, Name = World, i = 17, offset = 0
File opened, fp = 2199
Hello, World!
```

That's almost right, but the offset in the `sscanf` test should be `8`, not `0`.

suborb, 2023-02-07:
This was a puzzler for a bit because it was working last night when I committed, but not this morning.

Because you're using sdcc the format patterns can't be automatically detected, so you'll need to explicitly configure them with:

```
#pragma scan "%s %d %n"
```

The default selection works for most cases, but. `%n` isn't in that set!

imneme, 2023-02-13:
Ah, makes sense… Thanks.