z88dk/z88dk #2196: PHASE at linking time

suborb, 2023-02-20:
Some targets need to have code that accesses the graphics hardware located at a different address so that the code isn't paged out.

In classic, this achieves with the HIMEM section and a separate ORG directive.

Appmake automatically appends the HIMEM section to the main binary when creating the disk/tape file etc when -create-app is supplied as an argument.

However, for CP/M it might be preferred that a disk image isn't created and the generated .COM file is manually inserted into a different disk image. In this case -create-app isn't supplied and as a result the generated .COM file is missing the HIMEM section.

For example, compile examples/graphics/boxes with `zcc +cpm -subtype=pcw40 boxes.c` and see the two files created.

I was trying to produce a single binary, so I changed the `ORG __crt_org_graphics` in `lib/crt/classic/crt_section_standard.asm` to a `PHASE __crt_org_graphics`. However, all code within that section (which comes from library files) wasn't assembled to the PHASE address.

Is this expected/any way to achieve it (beyond always running -create-app)?

pauloscustodio, 2023-02-20:
The ```PHASE``` directive is handled at assembly-time only by replacing all address expressions into constants. The linker has no idea that a ```PHASE``` was in effect.

suborb, 2023-02-20:
That's a shame! I've updated the docs to indicate that so that I'd forget in future.

closed 2023-02-20 by suborb