z88dk/z88dk #2202: z88dk under WinXP + sources encoded in UTF-8 + BOM

Oleg-N-Cher, 2023-03-08:
I tried z88dk under Windows XP. It works great! I want to sincerely thank the developers that there is still support for this. The thing is, retro developers are very often retrograde, using old laptops or computers with old systems. And it's really great that z88dk is still working there!

But there is also one feature I want to let you know about. Several of my sources were UTF-8 + BOM encoded. You probably know this format. There are 3 bytes in the beginning - the BOM-signature. Such sources were perfectly built in z88dk version 2.2 under Windows 10 x64, but caused this error under Windows XP:

[image]

This is strange because the binaries of all z88dk utilities are the same (I tried 32-bit binaries from /bin.x86 folder under both Win10 and WinXP). Oh yes, it is built by SCCZ80 compiler for i8080-based target, if that's important.

suborb, 2023-03-08:
Just to confirm, the 32 bit binaries work correctly on Windows 10, but on Windows XP they trip over the BOM?

If that's the case, I'll add what I can only describe as a bodge to skip over the BOM if it's detected. 

BTW, If I've read the background window correctly, any enhancements for our PK-01 target (https://github.com/z88dk/z88dk/wiki/Platform---Lviv) would be gratefully received!

Oleg-N-Cher, 2023-03-09:
You got it right, **suborb**.

I suggest that you first make sure that the BOM problem is reproduced on XP at your side. After that, figure out exactly what mechanism misses the BOM on Win7-10, and figure out why it doesn't work on XP. Please don't make a stub.

I am not a Lviv PK-01 expert. Just helped one person to port my game [Bolder16K](https://github.com/Oleg-N-Cher/Bolder16K) to Lviv:

[image]

And also for Sura PK8000:

[image]

suborb, 2023-03-09:
>  BOM problem is reproduced on XP at your side

Yes, my first guess wasn't correct - I assumed that Windows 10 was swallowing the marker, XP wasn't and then sccz80 just didn't like the BOM. However, the BOM has no effect (and is consumed happily) on MacOS so looks like I'm off seeking an iso.

> And also for Sura PK8000:

Nice, I stalled a little on adding support for that one and never got as far as CP/M support.

suborb, 2023-03-09:
I've committed what was my original thought - just skip over the BOM. 

I've only applied it to sccz80 and it seems to work. However, there are many other places where we open files - if you run into any other problems just open another issue and we can generalise this workaround.

As to why it's breaking in the first place - I'm not sure - maybe a problem with `<ctype.h>`?

Oleg-N-Cher, 2023-03-10:
I can't know that, **suborb**.

The first thing I thought was that z88dk uses when opening a source code some internal mechanisms of the OS, which on Win7-10 knows how to skip BOM. The next thought made me doubt it - I thought it was unlikely you were using platform dependent mechanisms. Most likely, the code is written to be as portable as possible. So better you can see which mechanism skips BOM and why it doesn't do it on XP.

And does z88dk open the sources throughout the body of the compiler, and not in a unified way?

suborb, 2023-03-10:
> I can't know that, **suborb**.

I was just pondering - it wasn't meant to be an actual question!

> And does z88dk open the sources throughout the body of the compiler, and not in a unified way?

Each tool is independent of each other so will open the file itself.

suborb, 2023-03-19:
Aging out and closing since fix was pushed.

closed 2023-03-19 by suborb