z88dk/z88dk #2203: VG-5000 - ROM calls and interrupts

zx70, 2023-03-08:
The VG5000 used the interrupts to keep the video display refreshed.
Sadly the same interrupt code makes heavy use of the IX pointer in the exactly like the Sinclair BASIC does with IY.
Luckily Microsoft left a hook entry in RAM to intercept the interrupts, so I could use it to add a wrapper to save it.
@suborb, I'm not bold enough to revert the IXIY workaround, but I think it is now possible.

NOTE: the resulting code speed with interrupts enabled will be different depending on the ROM version, basing on the historical docs V1.0 is three times slower.

suborb, 2023-04-19:
From fdcdbbdc0bac16bdade9c0667038e9e5c0328d1d the interrupt is now extended and users can register interrupt routines with `add_raster_int()`

As an added bonus, -`pragma-define:CRT_DISABLE_FIRMWARE_ISR=0` will disable execution of the interrupt routine so timings can be a bit more deterministic.

zx70, 2023-04-19:
this is an absolutely valid option,  the earlier VG-5000 BASIC versions had a different, slower interrupt driven video handler

closed 2024-05-06 by suborb