z88dk/z88dk #2210: Create tek401x console + graphics driver

suborb, 2023-03-12:
Everyone loves an old school terminal. It used to be supported out of box by xterm on Linux/Solaris/Amiga etc etc These days we might have to build xterm specifically with it enabled.

"All" we need to do convert our graphics operations into output bytesand we can get some free-ish graphics for +test and other machines with just a serial port.

feilipu, 2023-03-13:
While you're at it, REGIS is pretty useful too. But also requires a "custom" xterm build.

It is fairly easy to do, as the language is well defined.

Some references here.

suborb, 2023-03-13:
Few glitches I need to figure out - might be a tekpoint/coordinate conversion issue but the random lines being printed is also odd.

[image]

suborb, 2023-03-14:
The problem was ticks doing some code remapping in it's output - fix coming along shortly.

It's not the most useful display, but I'm sure someone could find a use for it. I've stuck to the 4010 resolution of 1024x768 and moved the 0,0 coordinate to the top left corner to match other targets.

We've got a plot and a draw primitive. Plot uses point mode, draw uses graph mode so is a factor. circle() is hence software driven.

To use it, you need to define a function __tek_outc() which writes the character to the device. For +test/ticks this amounts to:

```
void __tek_outc(int c) __z88dk_fastcall
{
     fputc_cons(c);
}
```

For reference, `xterm -t` will open the Tektronix window

[image]

suborb, 2023-03-15:
> While you're at it, REGIS is pretty useful too, though not as high resolution. But it also requires a "custom" xterm build.
> 
> It is fairly easy to do, as the language is well defined.
> 
> Some references [here](https://github.com/feilipu/ReGIS).

And that'll be next on my list - seems a bit more useful than Tektronix since you've got more than just lines and pointss.

I'll probably lift your library, wrap it in classic graphics and have both APIs available.

feilipu, 2023-03-15:
> I'll probably lift your library, wrap it in classic graphics and have both APIs available.

In my confusion, I've linked in an Arduino library copy of the z80 version.
The [z88dk compliant REGIS version is here in my usual C library repo](https://github.com/feilipu/z88dk-libraries/tree/master/regis).

I've probably made a conceptual mistake by building "sentences" of instructions before committing to transmit them. That works ok in z80 world where there's a few kB of buffer available, but in AVR world with 2kB total RAM that doesn't work too well. In hindsight, it would probably be better to transmit as you go for z80 too.

I just did it with batched serial transmit to release my CPU to feed additional commands into the APU and have it work in parallel with the CPU serial interrupt sending the prior results, for some [planet projections](https://github.com/feilipu/planet-motion). But that's not a general case.

suborb, 2023-03-15:
> I've probably made a conceptual mistake by building "sentences" of instructions before committing to transmit them. That works ok in z80 world where there's a few kB of buffer available, but in AVR [...]

Yes, I wasn't sure about that so was probably going to unwind it - I think batching could be done at the `outc()` level if necessary.

suborb, 2023-03-16:
> Yes, I wasn't sure about that so was probably going to unwind it - I think batching could be done at the `outc()` level if necessary.

I have to admit I wasn't expecting the device to behave in the way it does - needing to enter and exit regis mode to have anything displayed makes things a bit awkward to fit in with `<graphics.h>` without having to provide a refresh() style function.

In progress: 

[image]

feilipu, 2023-03-17:
> needing to enter and exit regis mode to have anything displayed makes things a bit awkward...

Yes, REGIS is very much "sentence" orientated. In my code the sentence just needs to be written out to the serial port using `window_write()` (see [example here](https://github.com/feilipu/ReGIS/blob/main/examples/3d_arduino/3d_arduino.ino#L224), but this could be done code by code rather than in one go to avoid need for a command sentence buffer.

Then just doing a [`window_close()`](https://github.com/feilipu/z88dk-libraries/blob/master/regis/source/window_close.c) is all that is needed to finish the sentence and get it displayed on the screen. To continue with a new REGIS sentence it is begun with [`window_new()`](https://github.com/feilipu/z88dk-libraries/blob/master/regis/source/window_new.c).

But, I guess that is exactly what you mean by needing a `refresh()` function.

> In progress:

The progress looks good. Is it pixel by pixel? Perhaps it makes sense to do scan line based sentences? I don't know.

suborb, 2023-03-17:
It's been fun, but I think I'm done on this for now.  Both drivers support plotting and the primitives are supported with suitable delegation the hardware for line + circle. ReGIS also supports undrawing.

It's enough to get some of the graphics examples to run, but the lack of being able to read pixels limits the usability. But is perfect for drawing graphs and diagrams.

The blast from the past that is `pgplot` has been a quite useful resource, the ReGIS driver is here: https://lweb.cfa.harvard.edu/sma/miriad/wbcorrTest/downLoad/NewFormat/miriad-sma3.1.2/borrow/pgplot/drivers/vtdriv.f  and the Tektronix driver here: https://lweb.cfa.harvard.edu/sma/miriad/wbcorrTest/downLoad/NewFormat/miriad-sma3.1.2/borrow/pgplot/drivers/ttdriv.f

In the future, I'd like to add colour to the ReGIS support, but for the moment what we've got will do.

closed 2023-03-17 by suborb