z88dk/z88dk #2211: [einstein] Multicolor/other modes seemingly not working ?

gameblabla, 2023-03-14:
So i tried to play around with it and i only got mode 2 to work properly with the tatung einstein and the MSX library with set_mode(). (i thought it was supposed to be supported by the MSX library)
In fact set_mode() doesn't seem to do much on the tatung einstein.

I looked at https://github.com/z88dk/z88dk/blob/dbc48df496018ec3c3e0125a59c90b2686745e27/include/msx.h#L35
and i've only seen "magic" values for SVI, MSX and SC3000.

suborb, 2023-03-15:
* It looks like there's no code to switch to the multicolour mode 3
* The magic values should no longer be used, and instead 0,1,2 (3) should be used

Mode 0,1,2 should work though - I didn't get a chance to try it out last night but if you can share a problematic program I can take a look.

gameblabla, 2023-03-15:
0,1,2 do work yes, although i was most interested in mode 3 : (

suborb, 2023-03-15:
> 0,1,2 do work yes, (i checked again without using the defines) although i was most interested in mode 3 :(

Phew, okay, I'll try to get mode 3 added in the next couple of days - it ought to be there!

suborb, 2023-03-17:
Sorry, it took a bit longer than expected - ended up being sidetracked by old terminals! I've added in support for mode 3 and graphics is hooked up to it. Assuming it builds, the next nightly should have it.

Here's a very lores colour Mandelbrot: 

[image]

suborb, 2023-03-19:
As a heads up, I've got a branch that gets the "undocumented" mangled (mode 1+2) working on the Einstein and all other VDP targets.

closed 2023-03-19 by suborb