z88dk/z88dk #2215: SDCC sdcc Internal error: validateOpTypeConst failed in OP_VALUE(op) @ SDCCicode.c:1228: expected value, got symbol

Dark-Schneider-666, 2023-03-23:
Hi, I got this error when compiling setting SDCC as compiler:
`Internal error: validateOpTypeConst failed in OP_VALUE(op) @ SDCCicode.c:1228: expected value, got symbol`
The code is large with many files and don't know what could be, but I can put links to related:
[https://sourceforge.net/p/sdcc/bugs/3020/](url)
[https://sourceforge.net/p/sdcc/bugs/2983/](url)
[https://sourceforge.net/p/sdcc/bugs/2746/](url)
When checking version is the 4.2.0 that seems to be the latest, so probably is a SDCC bug itself that should be solved at origin.

Dark-Schneider-666, 2023-03-23:
OK I think I found the source.
At header:
```
#define graphics_base_address_registers ((GraphicsBuffersBaseAddressRegisters *)0xF866)

typedef struct {
	byte color_table_low_r3, color_table_high_r10 , pattern_generator_table_r4;
} SingleBufferBaseAddressRegisters;

typedef struct {
	byte pattern_layout_table_r2, sprite_attribute_table_low_r5, sprite_attribute_table_high_r11, sprite_pattern_generator_table_r6;
} DoubleBufferBaseAddresRegisters;

typedef struct {
	SingleBufferBaseAddressRegisters single_buffer;
	DoubleBufferBaseAddresRegisters double_buffer[2];
} BufferedBaseAddressRegisters;

typedef struct {
	byte gfx_tasks, working_buffer; // gfx_tasks to decide if swapbuffers; working_buffer is the current one for writing, the LSB is the index, so we can simply ++ it or +1 to get the visible buffer
	BufferedBaseAddressRegisters registers;
} GraphicsBuffersBaseAddressRegisters;
```
At C file:
```
void graphics_set_base_address_registers(BufferedBaseAddressRegisters *register_values) __z88dk_fastcall {
	memcpy(&graphics_base_address_registers->registers, register_values, sizeof(GraphicsBuffersBaseAddressRegisters));
	// reset state
	graphics_base_address_registers->gfx_tasks = 0;
	graphics_base_address_registers->working_buffer = 0;
	// put into registers
//	graphics_restore_base_address_registers();
}
```
The problem is at the memcpy for the &graphics_base_address_registers->registers parameter, it doesn't like with the & symbol, but is required to get the address. In zscc80 works fine.

Dark-Schneider-666, 2023-03-24:
OK seems SDCC has some problems to index addresses, it can be solved changing the & version by base address + offset manually.
As mentioned I suppose this is an issue to be solved at origin (SDCC).

closed 2023-03-24 by Dark-Schneider-666

feilipu, 2023-03-26:
See #2190.