z88dk/z88dk #2217: z88dk nightly build in macOS fails with error "error: non-const lvalue reference to type 'std::filesystem::path::iterator::reference' (aka 'std::filesystem::path') cannot bind to a temporary of type 'std::filesystem::path::iterator::reference' (aka 'std::filesystem::path')"

dbolli, 2023-04-06:
z88dk nightly build in macOS fails with error

c++ -std=gnu++17 -MMD -Wall -O3 -I. -Isrc -Wextra -Werror -pedantic-errors -Isrc/cpp -c -o src/cpp/utils.o src/cpp/utils.cpp
src/cpp/utils.cpp:262:13: error: non-const lvalue reference to type 'std::filesystem::path::iterator::reference' (aka 'std::filesystem::path') cannot bind to a temporary of type 'std::filesystem::path::iterator::reference' (aka 'std::filesystem::path')
        for (auto& elem : full_path) {
                   ^    ~
1 error generated.
make[1]: *** [src/cpp/utils.o] Error 1
make: *** [install] Error 2

Appears to be an issue with Xcode 14.3 as a previous nightly build from 2nd March 2023 that built successfully in Xcode 14.2 also now fails with the same error message

Temporarily fixed by restoring
  /usr/local/share/z88dk/
  /path/to/z88dk
using Time Machine

suborb, 2023-04-10:
Just updating one of my machines to investigate.

It does look like Xcode 14.3 has dropped a pile on a number of other projects, so let's see what mess it's created for us.

suborb, 2023-04-10:
Reopen in case of trouble...

closed 2023-04-10 by suborb

dbolli, 2023-04-13:
Re-tested successfully with Xcode 14.3 and nightly macOS build from 12/04/2023. Many thanks 👍