z88dk/z88dk #2218: Compiler error: op8_cost op: 7, aopGet got unsupported aop->type

tangent3D, 2023-04-06:
`FATAL Compiler Internal Error in file 'gen.c' line number '988' : code generator internal error` occurs when attempting to compile the following with `zcc +z80 test.c`:

`#define ARRAY_SIZE 5
void main()
{
    unsigned char array[ARRAY_SIZE];
    unsigned char* thisElement = array;
    unsigned char* nElement = array + ARRAY_SIZE;
    while(thisElement <  nElement) // does not work
  //while(thisElement != nElement) // works
    {
        *thisElement=0;
        thisElement++;
    }
}`

Console output is as follows:
`D:\>zcc +z80 test.c
op8_cost op: 7
test.c:11: error 9: FATAL Compiler Internal Error in file 'gen.c' line number '988' : code generator internal error
Contact Author with source code
op8_cost op: 7
op8_cost op: 7
op8_cost op: 7
(...)
aop->type: 7
test.c:11: error 9: FATAL Compiler Internal Error in file 'gen.c' line number '3072' : aopGet got unsupported aop->type
Contact Author with source code`

Issue occurs on 2023-04-06 win32 nighly build (z88dk-win32-20230406-c163a5749-20789).

xrrawva, 2023-04-06:
issue also occurs for https://github.com/z88dk/z88dk/releases z88dk-win32-2.2.zip

feilipu, 2023-04-06:
This is an SDCC bug that may have been fixed since our release using their 4.2.0.

Please see if you get the same error with our 4.3.0 branch.

If so then it can be reported on the SDCC bug tracker for action.

feilipu, 2023-04-06:
It may be resolved in their [#3503](https://sourceforge.net/p/sdcc/bugs/3503/). In which case please watch for their 4.3.0 release, which will trigger us to add it to our nightly build.

feilipu, 2023-04-06:
Unrelated, Z-Fighter looks very good.

tangent3D, 2023-04-08:
Thank you! 

I had a chance to build the 4.3.0 branch and the code above compiles without issue. Thanks for clearing this up @feilipu!

closed 2023-04-08 by tangent3D