z88dk/z88dk #2222: zcc: repeat loop dowhile undefined symbol i_13

RodrigoDornelles, 2023-04-10:
`:main::0::1:1: error: undefined symbol: i_13`

I use a lot of `do while` for repetition loops, and also for "jumping forward" instead of goto, and apparently it is not supported by the compiler.

```c
int main(int argc, char** argv)
{
  unsigned char result = 0;
  do {
     if (argc > 2) {
       result = 15;
       break;
     }
   }
   while(0);
   return result;
}
```

**PS.** am using latest commit cb918bc5d2d81f79285cbeb6710e73526077a97d 2023/04/10 18:00

RodrigoDornelles, 2023-04-10:
@suborb you were quick to solve the my last 2 previous problems, if you may be interested.

**Quickdraw**  :cowboy_hat_face:

suborb, 2023-04-11:
Obviously the minimal code example is a nonsense, but it should compile. I suspect what's happening is that while (0) is being evaluated to false and the terminal labels aren't being dropped. 

I'll look into it...

RodrigoDornelles, 2023-04-11:
> Obviously the minimal code example is a nonsense, but it should compile. I suspect what's happening is that while (0) is being evaluated to false and the terminal label isn't being dropped.
> 
> I'll look into it...

| [image] | I use minified code, because what I'm working on can be distracting due to its complexity.<br/><br/>basically, I did this to avoid `goto` and multiple return points. |
| :-: | :-- |

### legacy equivalent code (works) ###

basically here is a jump _(which is something outdated practice)_ to avoid multiple points of return.

```c
int main(int argc, char** argv)
{
   unsigned char result = 0;
   if (argc > 2) {
       result = 15;
       goto exit;
   }

   exit:
   return result;
}
```

### another equivalents code (works) ###

to check if it's problem with while(0), I did these two tests.

```c
int main(int argc, char** argv)
{
   unsigned char result = 0;
   do {
     if (argc > 2) {
       result = 15;
       break; ///< goto exit
     }
     break; ///< goto exit
   }
   while(1);
   return result;
}
```

```c
int main(int argc, char** argv)
{
   unsigned char result = 0;
   while(1) { 
     if (argc > 2) {
       result = 15;
       break; ///< goto exit
     }
     break; ///< goto exit
   }
   return result;
}
```

### expected code (not work)  ###

the code that I expected to work, basically it's this "goto forward" in a modern way, I've seen some embedded companies, it's an interesting practice.

```c
int main(int argc, char** argv)
{
  unsigned char result = 0;
  do {
     if (argc > 2) {
       result = 15;
       break; ///< goto exit
     }
   }
   while(0);
   return result;
}
```

an addendum I did a test with `if(0) {}` used as a practice of commenting code but leaving it visible from test coverage tools and it compiled correctly, this problem is just `do{ } while(0);` really.

suborb, 2023-04-11:
Related to #615

RodrigoDornelles, 2023-04-11:
done! 4b9c8711c70a4788226dd5bf18b7832cb170eb3a

closed 2023-04-11 by RodrigoDornelles