z88dk/z88dk #2223: link: Filetype of hello.c.obj unrecognized

RodrigoDornelles, 2023-04-11:
I don't know what it could be, it's compiled correctly, but fails when linking.

### when i use cmake & makefile

#### error

```
pastelzao@rodrigao:~/projects/3bc-lang$ make
Scanning dependencies of target tbc_sys_std
[  1%] Building C object CMakeFiles/tbc_sys_std.dir/src/sys/sys_common_mock.c.obj
[  3%] Building C object CMakeFiles/tbc_sys_std.dir/src/sys/sys_common_pexa.c.obj
[  3%] Built target tbc_sys_std
...
[ 39%] Built target tbc_array_simple
Scanning dependencies of target tbc_i18n
[ 40%] Building C object CMakeFiles/tbc_i18n.dir/src/i18n/i18n_no-ne.c.obj
[ 40%] Built target tbc_i18n
Scanning dependencies of target hello
[ 42%] Building C object CMakeFiles/hello.dir/programs/hello.c.obj
[ 43%] Linking C executable bin/hello.bin
Error running link command: No such file or directory
make[2]: *** [CMakeFiles/hello.dir/build.make:138: bin/hello.bin] Error 2
make[1]: *** [CMakeFiles/Makefile2:116: CMakeFiles/hello.dir/all] Error 2
make: *** [Makefile:84: all] Error 2
```

### when in run manually command

#### error
```
Filetype of CMakeFiles/hello.dir/programs/hello.c.obj unrecognized
```
#### hexdump

```bash
pastelzao@rodrigao:~/projects/3bc-lang$ hexdump -C CMakeFiles/hello.dir/programs/hello.c.obj
00000000  5a 38 30 52 4d 46 31 36  7e 03 00 00 1c 00 00 00  |Z80RMF16~.......|
00000010  85 01 00 00 63 03 00 00  b3 03 00 00 43 3e 00 2f  |....c.......C>./|
00000020  68 6f 6d 65 2f 70 61 73  74 65 6c 7a 61 6f 2f 70  |home/pastelzao/p|
00000030  72 6f 6a 65 63 74 73 2f  33 62 63 2d 6c 61 6e 67  |rojects/3bc-lang|
00000040  2f 70 72 6f 67 72 61 6d  73 2f 68 65 6c 6c 6f 2e  |/programs/hello.|
00000050  63 3a 3a 6d 61 69 6e 3a  3a 30 3a 3a 31 0f 00 00  |c::main::0::1...|
00000060  00 0d 00 63 6f 64 65 5f  63 6f 6d 70 69 6c 65 72  |...code_compiler|
00000070  00 00 01 00 00 00 14 00  5f 73 74 5f 6d 61 69 6e  |........_st_main|
00000080  5f 69 6e 73 74 61 6e 63  65 2b 34 35 43 00 00 0f  |_instance+45C...|
00000090  00 00 00 0d 00 63 6f 64  65 5f 63 6f 6d 70 69 6c  |.....code_compil|
000000a0  65 72 03 00 04 00 00 00  05 00 5f 70 72 6f 67 43  |er........_progC|
000000b0  00 00 0f 00 00 00 0d 00  63 6f 64 65 5f 63 6f 6d  |........code_com|
000000c0  70 69 6c 65 72 06 00 07  00 00 00 06 00 6c 5f 70  |piler........l_p|
000000d0  69 6e 74 43 00 00 0f 00  00 00 0d 00 63 6f 64 65  |intC........code|
000000e0  5f 63 6f 6d 70 69 6c 65  72 09 00 0a 00 00 00 14  |_compiler.......|
000000f0  00 5f 73 74 5f 6d 61 69  6e 5f 69 6e 73 74 61 6e  |._st_main_instan|
00000100  63 65 2b 34 37 43 00 00  0f 00 00 00 0d 00 63 6f  |ce+47C........co|
00000110  64 65 5f 63 6f 6d 70 69  6c 65 72 0e 00 0f 00 00  |de_compiler.....|
00000120  00 11 00 5f 73 74 5f 6d  61 69 6e 5f 69 6e 73 74  |..._st_main_inst|
00000130  61 6e 63 65 43 00 00 0f  00 00 00 0d 00 63 6f 64  |anceC........cod|
00000140  65 5f 63 6f 6d 70 69 6c  65 72 12 00 13 00 00 00  |e_compiler......|
00000150  11 00 5f 64 72 69 76 65  72 5f 69 6e 74 65 72 72  |.._driver_interr|
00000160  75 70 74 43 00 00 0f 00  00 00 0d 00 63 6f 64 65  |uptC........code|
00000170  5f 63 6f 6d 70 69 6c 65  72 18 00 19 00 00 00 03  |_compiler.......|
00000180  00 69 5f 33 00 4c 41 0f  00 72 6f 64 61 74 61 5f  |.i_3.LA..rodata_|
00000190  63 6f 6d 70 69 6c 65 72  00 00 00 00 05 00 5f 70  |compiler......_p|
000001a0  72 6f 67 32 00 2f 68 6f  6d 65 2f 70 61 73 74 65  |rog2./home/paste|
000001b0  6c 7a 61 6f 2f 70 72 6f  6a 65 63 74 73 2f 33 62  |lzao/projects/3b|
000001c0  63 2d 6c 61 6e 67 2f 70  72 6f 67 72 61 6d 73 2f  |c-lang/programs/|
000001d0  68 65 6c 6c 6f 2e 63 03  00 00 00 4c 41 0c 00 62  |hello.c....LA..b|
000001e0  73 73 5f 63 6f 6d 70 69  6c 65 72 00 00 00 00 11  |ss_compiler.....|
000001f0  00 5f 73 74 5f 6d 61 69  6e 5f 69 6e 73 74 61 6e  |._st_main_instan|
00000200  63 65 3e 00 2f 68 6f 6d  65 2f 70 61 73 74 65 6c  |ce>./home/pastel|
00000210  7a 61 6f 2f 70 72 6f 6a  65 63 74 73 2f 33 62 63  |zao/projects/3bc|
00000220  2d 6c 61 6e 67 2f 70 72  6f 67 72 61 6d 73 2f 68  |-lang/programs/h|
00000230  65 6c 6c 6f 2e 63 3a 3a  6d 61 69 6e 3a 3a 30 3a  |ello.c::main::0:|
00000240  3a 31 0f 00 00 00 4c 41  0d 00 63 6f 64 65 5f 63  |:1....LA..code_c|
00000250  6f 6d 70 69 6c 65 72 0e  00 00 00 03 00 69 5f 33  |ompiler......i_3|
00000260  3e 00 2f 68 6f 6d 65 2f  70 61 73 74 65 6c 7a 61  |>./home/pastelza|
00000270  6f 2f 70 72 6f 6a 65 63  74 73 2f 33 62 63 2d 6c  |o/projects/3bc-l|
00000280  61 6e 67 2f 70 72 6f 67  72 61 6d 73 2f 68 65 6c  |ang/programs/hel|
00000290  6c 6f 2e 63 3a 3a 6d 61  69 6e 3a 3a 30 3a 3a 31  |lo.c::main::0::1|
000002a0  0f 00 00 00 4c 41 0d 00  63 6f 64 65 5f 63 6f 6d  |....LA..code_com|
000002b0  70 69 6c 65 72 1b 00 00  00 03 00 69 5f 34 3e 00  |piler......i_4>.|
000002c0  2f 68 6f 6d 65 2f 70 61  73 74 65 6c 7a 61 6f 2f  |/home/pastelzao/|
000002d0  70 72 6f 6a 65 63 74 73  2f 33 62 63 2d 6c 61 6e  |projects/3bc-lan|
000002e0  67 2f 70 72 6f 67 72 61  6d 73 2f 68 65 6c 6c 6f  |g/programs/hello|
000002f0  2e 63 3a 3a 6d 61 69 6e  3a 3a 30 3a 3a 31 0f 00  |.c::main::0::1..|
00000300  00 00 47 41 0d 00 63 6f  64 65 5f 63 6f 6d 70 69  |..GA..code_compi|
00000310  6c 65 72 00 00 00 00 05  00 5f 6d 61 69 6e 3e 00  |ler......_main>.|
00000320  2f 68 6f 6d 65 2f 70 61  73 74 65 6c 7a 61 6f 2f  |/home/pastelzao/|
00000330  70 72 6f 6a 65 63 74 73  2f 33 62 63 2d 6c 61 6e  |projects/3bc-lan|
00000340  67 2f 70 72 6f 67 72 61  6d 73 2f 68 65 6c 6c 6f  |g/programs/hello|
00000350  2e 63 3a 3a 6d 61 69 6e  3a 3a 30 3a 3a 31 0f 00  |.c::main::0::1..|
00000360  00 00 00 06 00 6c 5f 70  69 6e 74 11 00 5f 64 72  |.....l_pint.._dr|
00000370  69 76 65 72 5f 69 6e 74  65 72 72 75 70 74 33 00  |iver_interrupt3.|
00000380  58 5f 68 6f 6d 65 5f 70  61 73 74 65 6c 7a 61 6f  |X_home_pastelzao|
00000390  5f 70 72 6f 6a 65 63 74  73 5f 33 62 63 5f 6c 61  |_projects_3bc_la|
000003a0  6e 67 5f 70 72 6f 67 72  61 6d 73 5f 68 65 6c 6c  |ng_programs_hell|
000003b0  6f 5f 63 18 00 00 00 0f  00 72 6f 64 61 74 61 5f  |o_c......rodata_|
000003c0  63 6f 6d 70 69 6c 65 72  ff ff ff ff 01 00 00 00  |compiler........|
000003d0  07 00 02 05 00 48 05 00  45 05 00 4c 05 00 4c 05  |.....H..E..L..L.|
000003e0  00 4f 05 00 21 05 00 0a  1f 00 00 00 0d 00 63 6f  |.O..!.........co|
000003f0  64 65 5f 63 6f 6d 70 69  6c 65 72 ff ff ff ff 01  |de_compiler.....|
00000400  00 00 00 11 00 00 21 00  00 cd 00 00 21 00 00 36  |......!.....!..6|
00000410  18 21 00 00 e5 cd 00 00  c1 7c b5 c2 00 00 21 00  |.!.......|....!.|
00000420  00 c9 65 01 00 00 0c 00  62 73 73 5f 63 6f 6d 70  |..e.....bss_comp|
00000430  69 6c 65 72 ff ff ff ff  01 00 00 00 00 00 00 00  |iler............|
00000440  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
*
000005a0  00 ff ff ff ff                                    |.....|
000005a5
```

suborb, 2023-04-11:
Paging @desertkun - I've never used cmake with z88dk

desertkun, 2023-04-11:
You have to explicitly specify a toolchain: https://github.com/z88dk/z88dk/wiki/CMake

```
cmake -DZCCTARGET=<target> -DCMAKE_TOOLCHAIN_FILE=/usr/local/share/z88dk/cmake/Toolchain-zcc.cmake <project>
```

Otherwise It would not use z88dk at all and go for regular gcc or whatever you have on your system.

RodrigoDornelles, 2023-04-12:
> You have to explicitly specify a toolchain: https://github.com/z88dk/z88dk/wiki/CMake
>
> ```
> cmake -DZCCTARGET=<target> -DCMAKE_TOOLCHAIN_FILE=/usr/local/share/z88dk/cmake/Toolchain-zcc.cmake [...]

I am using a toolchain for zcc, when i use `-DTARGET=<toolchain>-<plataform>` in my project.

https://github.com/RodrigoDornelles/3bc-lang/blob/develop-wip-wip/cmake/toolchain/zcc.cmake

RodrigoDornelles, 2023-04-12:
@desertkun 

## using cmake ##

#### commands ####

```
cmake . -DTARGET=zcc-sms -DCMAKE_VERBOSE_MAKEFILE=ON
make
```

#### full outputs ####

```
pastelzao@rodrigao:~/projects/3bc-lang$ cmake . -DTARGET=zcc-sms -DCMAKE_VERBOSE_MAKEFILE=ON
##################################
##              ##    ##        ##
##              ##    ##        ##
##              ##    ##    ######
############    ##    ##    ######
############    ##    ##    ######
##              ##    ##        ##
##              ##    ##        ##
##              ##    ############
############    ##              ##
############    ##              ##
##              ##    ######    ##
##              ##    ######    ##
##              ##              ##
##################################

-- Copyright (C) 2020 Rodrigo Dornelles.

-- using z88dk cross compiler.

-- Configuring done
-- Generating done
-- Build files have been written to: /home/pastelzao/projects/3bc-lang
pastelzao@rodrigao:~/projects/3bc-lang$ make
/usr/bin/cmake -S/home/pastelzao/projects/3bc-lang -B/home/pastelzao/projects/3bc-lang --check-build-system CMakeFiles/Makefile.cmake 0
/usr/bin/cmake -E cmake_progress_start /home/pastelzao/projects/3bc-lang/CMakeFiles /home/pastelzao/projects/3bc-lang/CMakeFiles/progress.marks
make -f CMakeFiles/Makefile2 all
make[1]: Entering directory '/home/pastelzao/projects/3bc-lang'
make -f CMakeFiles/tbc_sys_std.dir/build.make CMakeFiles/tbc_sys_std.dir/depend
make[2]: Entering directory '/home/pastelzao/projects/3bc-lang'
cd /home/pastelzao/projects/3bc-lang && /usr/bin/cmake -E cmake_depends "Unix Makefiles" /home/pastelzao/projects/3bc-lang /home/pastelzao/projects/3bc-lang /home/pastelzao/projects/3bc-lang /home/pastelzao/projects/3bc-lang /home/pastelzao/projects/3bc-lang/CMakeFiles/tbc_sys_std.dir/DependInfo.cmake --color=
Scanning dependencies of target tbc_sys_std
make[2]: Leaving directory '/home/pastelzao/projects/3bc-lang'
make -f CMakeFiles/tbc_sys_std.dir/build.make CMakeFiles/tbc_sys_std.dir/build
make[2]: Entering directory '/home/pastelzao/projects/3bc-lang'
[  1%] Building C object CMakeFiles/tbc_sys_std.dir/src/sys/sys_common_mock.c.obj
export ZCCCFG=/home/pastelzao/projects/z88dk/lib/config && export PATH=/home/pastelzao/projects/z88dk/bin:$PATH && /home/pastelzao/projects/z88dk/bin/zcc +sms   -I/home/pastelzao/projects/3bc-lang/src -I/home/pastelzao/projects/3bc-lang/src/bus -I/home/pastelzao/projects/3bc-lang/src/cpu -I/home/pastelzao/projects/3bc-lang/src/ds -I/home/pastelzao/projects/3bc-lang/src/pkg -I/home/pastelzao/projects/3bc-lang/src/sys -I/home/pastelzao/projects/3bc-lang/src/i18n -I/home/pastelzao/projects/3bc-lang/src/types -I/home/pastelzao/projects/3bc-lang/src/interpreter   +sms   -o CMakeFiles/tbc_sys_std.dir/src/sys/sys_common_mock.c.obj -c /home/pastelzao/projects/3bc-lang/src/sys/sys_common_mock.c
[  3%] Building C object CMakeFiles/tbc_sys_std.dir/src/sys/sys_common_pexa.c.obj
export ZCCCFG=/home/pastelzao/projects/z88dk/lib/config && export PATH=/home/pastelzao/projects/z88dk/bin:$PATH && /home/pastelzao/projects/z88dk/bin/zcc +sms   -I/home/pastelzao/projects/3bc-lang/src -I/home/pastelzao/projects/3bc-lang/src/bus -I/home/pastelzao/projects/3bc-lang/src/cpu -I/home/pastelzao/projects/3bc-lang/src/ds -I/home/pastelzao/projects/3bc-lang/src/pkg -I/home/pastelzao/projects/3bc-lang/src/sys -I/home/pastelzao/projects/3bc-lang/src/i18n -I/home/pastelzao/projects/3bc-lang/src/types -I/home/pastelzao/projects/3bc-lang/src/interpreter   +sms   -o CMakeFiles/tbc_sys_std.dir/src/sys/sys_common_pexa.c.obj -c /home/pastelzao/projects/3bc-lang/src/sys/sys_common_pexa.c
make[2]: Leaving directory '/home/pastelzao/projects/3bc-lang'
[  3%] Built target tbc_sys_std
make -f CMakeFiles/tbc_legacy.dir/build.make CMakeFiles/tbc_legacy.dir/depend
make[2]: Entering directory '/home/pastelzao/projects/3bc-lang'
cd /home/pastelzao/projects/3bc-lang && /usr/bin/cmake -E cmake_depends "Unix Makefiles" /home/pastelzao/projects/3bc-lang /home/pastelzao/projects/3bc-lang /home/pastelzao/projects/3bc-lang /home/pastelzao/projects/3bc-lang /home/pastelzao/projects/3bc-lang/CMakeFiles/tbc_legacy.dir/DependInfo.cmake --color=
Scanning dependencies of target tbc_legacy
make[2]: Leaving directory '/home/pastelzao/projects/3bc-lang'
make -f CMakeFiles/tbc_legacy.dir/build.make CMakeFiles/tbc_legacy.dir/build
make[2]: Entering directory '/home/pastelzao/projects/3bc-lang'
[  4%] Building C object CMakeFiles/tbc_legacy.dir/src/legacy/driver_memory.c.obj
export ZCCCFG=/home/pastelzao/projects/z88dk/lib/config && export PATH=/home/pastelzao/projects/z88dk/bin:$PATH && /home/pastelzao/projects/z88dk/bin/zcc +sms   -I/home/pastelzao/projects/3bc-lang/src -I/home/pastelzao/projects/3bc-lang/src/bus -I/home/pastelzao/projects/3bc-lang/src/cpu -I/home/pastelzao/projects/3bc-lang/src/ds -I/home/pastelzao/projects/3bc-lang/src/pkg -I/home/pastelzao/projects/3bc-lang/src/sys -I/home/pastelzao/projects/3bc-lang/src/i18n -I/home/pastelzao/projects/3bc-lang/src/types -I/home/pastelzao/projects/3bc-lang/src/interpreter   +sms   -o CMakeFiles/tbc_legacy.dir/src/legacy/driver_memory.c.obj -c /home/pastelzao/projects/3bc-lang/src/legacy/driver_memory.c
[  6%] Building C object CMakeFiles/tbc_legacy.dir/src/legacy/ds_hypervisor_darray.c.obj
export ZCCCFG=/home/pastelzao/projects/z88dk/lib/config && export PATH=/home/pastelzao/projects/z88dk/bin:$PATH && /home/pastelzao/projects/z88dk/bin/zcc +sms   -I/home/pastelzao/projects/3bc-lang/src -I/ho [... cut ...] er_power.c.obj CMakeFiles/tbc_driver_no_math.dir/src/driver/driver_program.c.obj CMakeFiles/tbc_driver_no_math.dir/src/driver/driver_tty.c.obj CMakeFiles/tbc_array_simple.dir/src/ds/ds_prog_array.c.obj CMakeFiles/tbc_array_simple.dir/src/ds/ds_ram_array.c.obj CMakeFiles/tbc_bus_cpu_hello.dir/src/bus/bus_cpu_hello.c.obj CMakeFiles/tbc_bus_sys_hello.dir/src/bus/bus_sys_hello.c.obj -o bin/hello.bin 
```

### fulloutputs ###

```
Filetype of CMakeFiles/hello.dir/programs/hello.c.obj unrecognized
```

desertkun, 2023-04-12:
To build files that end win non-c extention, you need to additionally specify `-xc` argument.

suborb, 2023-04-12:
The error is coming from zcc - its not recognising .obj files as object files.

desertkun, 2023-04-12:
I believe the correct solution is to force CMake to generate `.o` object files.

```
set(CMAKE_USER_MAKE_RULES_OVERRIDE "${CMAKE_CURRENT_LIST_DIR}/Extensions.cmake" CACHE STRING "" FORCE)
```

Extensions.cmake:
```
set(CMAKE_C_OUTPUT_EXTENSION ".o")
set(CMAKE_ASM_OUTPUT_EXTENSION ".o")
```

suborb, 2023-04-15:
I think this one has been sorted with both zcc handling .obj files and the cmake recipe above.

closed 2023-04-15 by suborb