z88dk/z88dk #2229: Progressing with the 8080 libraries

zx70, 2023-04-21:
I tried to extend the library coverage for the non-Z80 CPUs.

--- Building +z80 (8080) Library ---

qsort_sccz80.asm:20: error: illegal identifier
  ^---- ld (_jp_qsort_compare+1),hl

What's wrong ?

I declared it in another source file, in this way

IF __CPU_GBZ80__ || __CPU_INTEL__

SECTION smc_clib
	PUBLIC _jp_qsort_compare

._jp_qsort_compare
   jp 0	; JP <compare>

ENDIF

pauloscustodio, 2023-04-21:
The gbz80 does not support the instruction ```ld (%m),hl``` Is this the case?

pauloscustodio, 2023-04-21:
Note: https://raw.githubusercontent.com/z88dk/z88dk/master/src/z80asm/dev/cpu/opcodes.txt contains all the valid opcodes recognized by z80asm

zx70, 2023-04-21:
I tested the 8085 but forgot the game boy, so thankyou for suggesting.  By the way I'm on the 8080 yet.
It seems to be accepting the opcode but fails at the linking stage, while gathering the label values.

zx70, 2023-04-28:
Stuck again.  It builds now, but when I try to use the new functions, they apparently are missing.

    othello.c::my_mov::0::38:835: error: undefined symbol: qsort_sccz80_callee
      ^---- qsort_sccz80_callee

I edited the Makefile and removed the exclusions for *sort*, what else should I do ?

    ASMFILES_8080 = $(filter-out $(wildcard *sqrt*.asm *inp*.asm *outp*.asm *extract*.asm) , $(wildcard *.asm)) $(wildcard 8080/*.asm)
    ASMFILES_GBZ80 = $(filter-out $(wildcard *sqrt*.asm *inp*.asm *outp*.asm *extract*.asm) , $(wildcard *.asm)) $(wildcard 8080/*.asm)