z88dk/z88dk #2254: (z80asm) "Not an object file"

suborb, 2023-05-08:
Given the length of time it takes to build libsrc these days I'm investigating getting "make -j N" to work correctly.

I've done a fair bit of cleanup of the main Makefile to remove obvious blockers, however I'm running into an issue which I think is caused by z80asm attempting to append/check object files.

Because we just a recursive make (sorry), instantiating -j N at the top level will end up with multiple instances of make visiting the subdirectories.

Effectively we end up with a bit of race condition, where end up with two instances trying to assemble the same file. If you're very unlucky you can end up in a situation where the following happens:

```
make -j 4 Q=

z88dk-z80asm -I../ -I/Users/dom/z88dk/libsrc//../lib/ -I/Users/dom/z88dk/libsrc//../libsrc/ -IXIY -mz80 -D__CLASSIC  -Oobj/ixiy __scanf_consume_whitespace.asm
error: not an object file: obj/ixiy/__scanf_common_start.o
error: file open: obj/ixiy/__scanf_common_start.o
make[1]: *** [obj/ixiy/__scanf_common_start.o] Error 1
```

Perhaps we need to move the object file into place once it's been completely generated or open with an exclusive read/write lock to avoid this problem.

pauloscustodio, 2023-05-13:
@suborb, I have changed z80asm to create a temporary object file and rename after generation. There is still the possibility of a race condition in zobjfile.c line 269 between these two lines:
```
	remove(obj_filename);
	int rv = rename(utstring_body(temp_filename), obj_filename);
```
The remove is needed because rename fails if the file exists.

closed 2023-05-13 by pauloscustodio

pauloscustodio, 2023-05-15:
> assuming that the temporary file is in the same directory as the output file.
Yes, they are

commit d5e96f20 (2023-05-13): Fix #2254: move the object file into place once it's been generated