z88dk/z88dk #2258: (math32) Linking issues + general confusion

suborb, 2023-05-12:
I'm a little confused as to what's happened with math32 - as far as I can tell bits of the library can't work with sdcc/classic and the same functions also cause trouble with linking.

```
#include <math.h>
#include <stdio.h>

#pragma printf "%f"

int main()
{
        double y;
        double x = fmod(11, 2);
        printf("%lf\n",x);
}
```

Fails to link (see https://z88dk.org/forum/viewtopic.php?f=2&t=11786)

Correcting, so that the SDCC function is exported as defc _fmod_callee = fmod_callee then the result is wrong.

A bit of head scratching shows the header file:

```
#ifdef __SCCZ80
extern double_t __LIB__ fmod(double_t x,double_t y) __smallc;
extern double_t __LIB__ fmod_callee(double_t x,double_t y) __smallc __z88dk_callee;
#define fmod(a,b) fmod_callee(a,b)
#endif

#ifdef __SDCC
extern double_t fmod(double_t x,double_t y);
extern double_t fmod_callee(double_t x,double_t y) __z88dk_callee;
#define fmod(a,b) fmod_callee(a,b)
#endif
```

So the calling conventions differ which isn't how classic works.

```
find lm32/c/sccz80 | xargs grep sdcc
```

Shows a lot of references to sdcc functions which isn't what I was expecting.

So I'm really confused and I've missed something, @feilipu can you remember and enlighten me?

feilipu, 2023-05-13:
IIRC, the main issue is that most of the library is written in assembly, and but some functions are written in C [and pre-compiled by SDCC_IX](https://github.com/z88dk/z88dk/blob/master/libsrc/_DEVELOPMENT/math/float/math32/c/Makefile#L10) and linked as assembly. So sometimes there are native labels, and sometimes the linkage has to deal with SDCC produced labels.

So that resulted in quite a confusing mix of native z80 assembly and SDCC labels being used, depending on where the function originated. And that is why you are seeing SDCC label references.

Compounding that, my testing was being done mainly on newlib targets (`sdcc_ix`, `sdcc_iy`, and sccz80 `new`) means that classic+SDCC combined only got `test` target testing, or possibly not tested at all.

Unfortunately this issue (classic+SDCC) might also flow onto `am9511` and `float16` libraries, as I would have done testing the same way there too. Can dig into that later.

---

Side note. I've been dealing with a vision issue over the past few weeks, culminating in a [Virectomy](https://www.asrs.org/patients/retinal-diseases/25/vitrectomy) last week so, my apologies but, I'm not going to be able to contribute to fixing this for a little while.

feilipu, 2023-05-13:
Thinking about this further, quite a few things have changed since `math32` was written, `newlib` is now being integrated into classic, and I've got a better handle on things now too. So some suggestions for discussion.

I think it would be better if the `math32` and `am9511` C functions were compiled with sccz80 across the board for z80 targets. It is done this way with [`am9511_8085`](https://github.com/z88dk/z88dk/blob/master/libsrc/_DEVELOPMENT/math/float/am9511/c/Makefile#L14) and [`math16`](https://github.com/z88dk/z88dk/blob/master/libsrc/_DEVELOPMENT/math/float/math16/c/Makefile#L10) libraries, but it should be generalised. That way use of the index registers could be completely avoided, making `math32` more useful for classic z80 targets that use the index registers for interrupts, or video, etc. And this would align `math32` more closely with my other two maths libraries too.

I'm not sure how best to handle the intermediate binding files more sensibly. At the moment there's a flood of files, that need to be generated individually. Somehow the number of different cases (compiler, library, target type) needs to be rationalised so that linkage for 5 different cases is minimised, `sdcc_iy`, `sdcc_ix`, sccz80 `new`, sccz80 `classic`, and sdcc `classic`.

Comments or follow on thoughts welcome.

suborb, 2023-05-13:
Thanks for the info - it sounds like the breakage of the classic side of things was an oversight so I will repair it.

I had a similar explosion when doing fix16 (https://github.com/z88dk/z88dk/tree/master/libsrc/math/fix16/c/sccz80) and like the other maths libraries it's very formulaic so a prime candidate for automated generation of the stubs.

The only real area of complexity is the multi-parameter functions where the actual implementation uses a different calling convention to the stub and we have to swap parameters around. For these I wonder if it's worth while compiling those files twice: once with `__smallc` and once with `__stdc` and always using the `__smallc` variants for classic + newlib-sccz80 and the `__stdc` for newlib-sdcc. This way we avoid having things like cm32_sccz80_fmod_callee.asm which are just interposing stack shufflers. I'll have a bit more of a think about it.

Removing the use of ix isn't that important - I think there's only a couple of targets that reserve both index registers and there's plenty of other library code that uses ix (long functions for example). 

I'm fairly confident that am9511 got hit by the same problem, so I'll fix up the classic math32 sdcc bridges, extend the test suite to cover more functions and see what falls out with am9511.

No worries about not being able to untangle, it shouldn't take too long to do and your input has been invaluable. Hopefully your vision won't take too long to recover.

suborb, 2023-05-13:
I've got a commit pending for fixing math32/sccz80/classic.

But, I did find some problems with am9511:

* The `__callee` functions are broken for sccz80 - there's an extra return address on the stack before the parameters
* The cimpl of these functions need to be marked as `__stdc` on 8085 since they are compiled with sccz80 or the argument swapping is incorrect.
* `cam32_sccz80_modf_callee` assumes that both arguments are 4 bytes 

I'm tempted to remove the callee annotation for am9511 as quickish fix.

feilipu, 2023-05-14:
Ok do that as it will be a few weeks before I could look into it.
Thank you.

Perhaps as a separate commit to make it easier to see what. needs to be done restore `_callee` later.

suborb, 2023-05-14:
Some other problems with am9511:

- [x] fmin/fmax jump to asm_sccz80_read* so the stack offset is incorrect
- [ ] 8085 asm_am9511_compare_sccz80 mutates the passed in values which breaks fmin and fmax
- [x] z80 l_long_div* l_long_mod* use the exx set which breaks classic printing (which assumes the exx set is untouched)

feilipu, 2023-05-16:
-[] 8085 `asm_am9511_compare_sccz80` mutates the passed in values which breaks `fmin` and `fmax`.

I can (now) see the issue, and I think I'd resolve it by replication of the parameters on the stack before comparing, using a `ldir` equivalent.

The weekend will be good for 👀.

Unless you get to it first.

feilipu, 2023-05-16:
You're right. Probably no less efficient to use cimpl C and the "fast" comparison code just as it stands.

feilipu, 2023-05-22:
Think this could be closed done, given no additional forum feedback.

closed 2023-05-22 by suborb