z88dk/z88dk #2259: Build of zcc fails due to missing BINDIR

pauloscustodio, 2023-05-13:
```
cc -c -o zcc.o -O2 -DLOCAL_REGEXP -Wall -pedantic -g -MMD -I. -I../copt -I../common -I../../ext/uthash/src zcc.c
zcc.c: In function 'main':
zcc.c:919:41: error: 'BINDIR' undeclared (first use in this function); did you mean 'EISDIR'?
  919 |     if ( strcmp(ptr,".") && strcmp(ptr, BINDIR)) {
      |                                         ^~~~~~
      |                                         EISDIR

```

pauloscustodio, 2023-05-13:
Error was caused by incomplete src/config.h:
```
#define PREFIX "/usr/local/share/z88dk"
#define Z88DK_VERSION "MSC build-Sat 08/20/2022"
```
and main Makefile says 'config.h' is up-to-date.

Who creates the incomplete config.h?

feilipu, 2023-05-13:
`src/config.h` is not cleaned, so an old version persists in "used" repos. 

https://github.com/z88dk/z88dk/issues/2226#issuecomment-1528927317

closed 2023-05-15 by pauloscustodio