z88dk/z88dk #2264: No autocast result to function definition

Dark-Schneider-666, 2023-05-16:
I have this definitions:
`typedef uint8_t bool;`
`extern bool collide_check_u16(uint16_t x, uint16_t y, Rectangle_u16 *collision) __smallc __z88dk_callee;`
The returning values is like:
```
collide_return_false:
    ld l, 0
    ret
collide_return_true:
    ld l, 1
    ret
```
When doing something like:
`collide_draw_set(collide_check_u16(100, 90, &rect2) ? 8 : 7);`
It uses also the value at register H, so getting incorrect behavior, if instead I do:
`collide_draw_set((bool)collide_check_u16(100, 90, &rect2) ? 8 : 7);`
Works normally, using only register L.

suborb, 2023-05-22:
I think this is another manifestation of #2216

Dark-Schneider-666, 2023-05-23:
I am using SmallC, should have indicated at title.

suborb, 2023-05-23:
Ah, in that case it's working as expected: https://github.com/z88dk/z88dk/wiki/CallingConventions - return values with `__smallc` are in hl, or dehl

Dark-Schneider-666, 2023-05-23:
Then I have to revise some of my asm code.

Thanks.

closed 2023-05-23 by Dark-Schneider-666