z88dk/z88dk #2269: [feature/question] Very low max UDG for targets with hardware support for 128 UDGs

Fabrizio-Caruso, 2023-05-18:
In features.h I see that Jupiter Ace and EG2000 targets in Z88DK only support 32 UDGs and Sorcerer just 16 UDGs.
From what I have read all of these targets have hardware support for 128 UDGs.

Could Z88DK increase the number of supported UDGs to a higher value (e.g. 64 or more)?

Is there any specific technical reasons why (only) these targets currently have such a low number of UDGs?

suborb, 2023-05-21:
On +ace the font needs to be copied into RAM and since there are only 128 slots available we’re down to 32 UDGs. 

The other two do support 128 udgs as long as you don’t use a custom font. 

Unfortunately this info isn’t encapsulated on features.h. So a couple of extra columns are needed - custom font available and udgs available when custom font used. 

Hopefully with that added you‘ll be able to take a compile time choice whether to use a font with embedded graphics or a separate udg bank. 

Let me know if that‘s enough to help you and I’ll add them.

suborb, 2023-05-21:
I've updated, so there's now 3 properties:

```
#define __GENCON_CUSTOM_FONT
#define __GENCON_MAX_UDGS
#define __GENCON_MAX_UDGS_WITH_CUSTOM_FONT
```

Which should (with a bit of logic) allow you to determine what assets to compile.  There's only a handful of targets where `__GENCON_MAX_UDGS` != `__GENCON_MAX_UDGS_WITH_CUSTOM_FONT` which are:  +cpm --subtype=bee, +bee, +trs80 --subtype=eg2000disk, +srr, +super80 --subtype=vduem. And of course +ace just has a low number available.

Fabrizio-Caruso, 2023-05-21:
@suborb Thanks a lot! In Cross-Lib's abstract model, all UDGs are statically fixed at compile time because I cannot assume all targets (not just Z88DK ones) can re-define UDGs at run-time.
So using custom fonts to statically set UDGs at start-up is fine with what I need to do.
I don't know whether I have not used custom fonts or UDGS in another in Z88DK, though. 
I have used this:
https://github.com/Fabrizio-Caruso/CROSS-LIB/blob/master/src/cross_lib/display/init_graphics/z88dk/udg/udg_init_graphics.c#L11

I supposed I need to use something else to access more than 32 UDGs.

How do I use custom fonts to get the highest possible number of UDGs?

zx70, 2023-05-23:
This is a probably unrelated question, but it's still on the UDG subject.

I just implemented the FONTs and UDGs on the Visual 1050.
As for the Microbee and probably others (x1 ?)  I framed the 8x8 resources in the 8x12 character cell.
I know we have bigger fonts too:  what's the approach you chose ?  More cross-compatibility or target specific features first ?

Fabrizio-Caruso, 2023-05-23:
@suborb This is very nice. Indeed having both is the perfect solution.
For Cross-Lib I do support several different tile shapes because I need to cover all pixels, otherwise things like vertical walls would just show empty lines.
In particular I currently support 
ASCII tiles for targets with no graphics, 8x8 for nearly all targets with graphics, 6x8 (for Oric and some RCA 1802 systems), 6x9 (for most RCA 1802 systems), 8x6 (for Atari Lynx console), 7x8 (for the Apple ][ series). I plan to add 8x12 and 8x10.
My scripts can automatically stretch or extend 8x8 tiles into other shapes but for the best results I redraw them in my games.
So, my code is 100% abstract but the assets may need some adaptations for the best results and I re-use assets of the same size across several targets.

Fabrizio-Caruso, 2023-05-29:
@suborb @zx70 what does custom font UDGs vs non-costom font UDGs mean?
I suppose I am using custom fonts here:
https://github.com/Fabrizio-Caruso/CROSS-LIB/blob/master/src/cross_lib/display/init_graphics/z88dk/udg/udg_init_graphics.c#L11

How can I define more UDGs in a different way? How?
Are there any examples?

suborb, 2023-05-29:
UDGS are by convention character codes >= 128

If you can define a custom font (ascii codes 32->127) then you can of course redefine any character with that range that you don't need for displaying the UI.

Fabrizio-Caruso, 2023-05-29:
@suborb Practically what do I need to do to be able to define more than 32 characters?
In my code I use IOCTL_GENCON_SET_FONT32
Is this limiting me to 32 redefined characters?
If yes, what do I need to use?

suborb, 2023-05-29:
For the EG2000 and Sorcerer just call IOCTL_GENCON_SET_UDGS with 128 UDGS *do not* call IOCTL_GENCON_SET_FONT.

For +ace, you'll need to intermix the graphics with standard font glyphs using IOCTL_GENCON_SET_FONT *and* call IOCTL_GENCON_SET_UDGS with a bank of 32 UDGs.

closed 2023-07-08 by suborb