z88dk/z88dk #2278: Agon Light is not supported yet

snakebyte69, 2023-05-28:
https://www.thebyteattic.com/2022/04/announcing-agon.html#!

suborb, 2023-05-31:
It's a start:

[image]

suborb, 2023-06-02:
Status (in agon_support branch):

* [x] crt0
* [x] Firmware printing
* [x] Gencon with colours + fonts UDGs
* [x] `vpeek()` with regular font
* [ ] `vpeek()` with UDGS (needs a VDP change, see https://github.com/breakintoprogram/agon-vdp/issues/56)
* [x] `fgetc_cons()`
* [x] `getk()`
* [ ] `plot()` (in progress)
* [ ] `unplot()` (would be better if `VDU 25, 0x4*3*, ...`was implemented)
* [ ] `xorplot()` (really need `VDU 18, *3*, col` to do this properly)
* [ ] `pointxy()` (can be done)
* [ ] Accelerated `draw()/undraw()`
* [ ] Not-accelerated `xordraw()`
* [ ] Accelerated `drawb()/undrawb()`
* [ ] Not-accelerated `xordrawb()`
* [ ] Accelerated `circle()/uncircle()`
* [ ] Not-accelerated `xorcircle()`
* [ ] Keyboard based `joystick()`

suborb, 2023-06-06:
I'm going to call this one done for the moment, at least until I can test it on real hardware.