z88dk/z88dk #2288: param --debug not working

dgzornoza, 2023-06-03:
I try to test samples for zx spectrum in doc with -debug param, but it's not working.
Steps:

create sample source file:
```
#include <arch/zx.h>

int main()
{
    zx_border(INK_BLACK);
    return 0;
}
```

try to compile:
it's works
`zcc +zx -clib=sdcc_iy -m -v -create-app main.c`
but with -debug info not works
`zcc +zx -clib=sdcc_iy -m -v -debug -create-app main.c`

error:
`cat: can't open '/tmp/tmpXXHGEPEK.adb': No such file or directory`

log with verbose command:

```
PROCESSING main.c
z88dk-zsdcpp -iquote"." -D__Z88DK -D__SPECTRUM -DSPECTRUM -D__SPECTRUM__ -D__Z80 -DZ80 -D__SDCC -D__SDCC_IY -isystem"/opt/z88dk/lib/config//../..//include/_DEVELOPMENT/sdcc"  -D__SDCC  "main.c" "/tmp/tmpXXgkfAnc.i2"
z88dk-zpragma -zcc-opt="/tmp/tmpzccXXpGhJoc/zcc_opt.def" < "/tmp/tmpXXgkfAnc.i2" > "/tmp/tmpXXgkfAnc.i"
z88dk-zsdcc --constseg rodata_compiler   --reserve-regs-iy -mz80 --no-optsdcc-in-asm --c1mode --emit-externs  --no-c-code-in-asm   --debug --no-peep --peep-file "/opt/z88dk/lib/config//../..//libsrc/_DEVELOPMENT/sdcc_peeph.2"  < "/tmp/tmpXXgkfAnc.i" -o "/tmp/tmpXXgkfAnc.opt"
z88dk-copt -mz80  "/opt/z88dk/lib/config//../..//libsrc/_DEVELOPMENT/sdcc_opt.1" < "/tmp/tmpXXgkfAnc.op1" > "/tmp/tmpXXgkfAnc.opt"
z88dk-copt -mz80  "/opt/z88dk/lib/config//../..//libsrc/_DEVELOPMENT/sdcc_opt.9" < "/tmp/tmpXXgkfAnc.opt" > "/tmp/tmpXXgkfAnc.op1"
z88dk-copt -mz80  "/opt/z88dk/lib/config//../..//libsrc/_DEVELOPMENT/sdcc_opt.2" < "/tmp/tmpXXgkfAnc.op1" > "/tmp/tmpXXgkfAnc.asm"
cat "/tmp/tmpXXgkfAnc.asm" >> "/tmp/tmpXXgkfAnc.tmp"
cat "/tmp/tmpXXgkfAnc.tmp" >> "/tmp/tmpXXgkfAnc.asm"
z88dk-z80asm -I"/opt/z88dk/lib/config//../..//lib/target/zx/def" -D__SDCC_IY -I"/opt/z88dk/lib/config//../..//libsrc/_DEVELOPMENT/target/zx" -m -debug -I"/tmp/tmpzccXXpGhJoc" -D__SDCC  -s  -mz80 -I"/opt/z88dk/lib/config//../..//lib" "/tmp/tmpXXgkfAnc.asm"

PROCESSING /opt/z88dk/lib/config//../..//libsrc/_DEVELOPMENT/target/zx/zx_crt.asm.m4
m4 -I "/opt/z88dk/lib/config//../..//libsrc/_DEVELOPMENT/target/zx" -I "/tmp/tmpzccXXpGhJoc" -I "/opt/z88dk/lib/config//../..//src/m4"   < "/opt/z88dk/lib/config//../..//libsrc/_DEVELOPMENT/target/zx/zx_crt.asm.m4" > "/tmp/tmpXXGFomoc.asm"
cat "/tmp/tmpXXGFomoc.asm" >> "/tmp/tmpXXGFomoc.tmp"
cat "/tmp/tmpXXGFomoc.tmp" >> "/tmp/tmpXXGFomoc.asm"
WILL ACT AS CRT

GENERATING OUTPUT

PROCESSING CRT
z88dk-z80asm -I"/opt/z88dk/lib/config//../..//lib/target/zx/def" -D__SDCC_IY -I"/opt/z88dk/lib/config//../..//libsrc/_DEVELOPMENT/target/zx" -m -debug -I"/tmp/tmpzccXXpGhJoc" -D__SDCC  -s  -mz80 -I"/opt/z88dk/lib/config//../..//lib" "/tmp/tmpXXGFomoc.asm"

z88dk-z80asm -mz80 -b -d  -o"a.bin" -m -s -L.  -L"/opt/z88dk/lib/config//../..//libsrc/_DEVELOPMENT/lib/sdcc_iy"   -D__SDCC_IY -I"/opt/z88dk/lib/config//../..//libsrc/_DEVELOPMENT/target/zx" -debug -I"/tmp/tmpzccXXpGhJoc" -D__SDCC -lzx    "/tmp/tmpXXGFomoc.o" "/tmp/tmpXXgkfAnc.o"
cat "/tmp/tmpXXGFomoc.adb" >> "a.adb"
cat: can't open '/tmp/tmpXXGFomoc.adb': No such file or directory
z88dk-appmake +zx  -b "a.bin" -c "a"
```

the file a.adb is not created and can not debug.

¿not working -debug param?
y try to use `-no-cleanup` but nothing happens.

thanks

desertkun, 2023-06-03:
Does it work without clib argument?
I think the reason is, only sccz80 might have the debug support unfortunately, @suborb correct me if I'm wrong?

suborb, 2023-06-03:
I couldn't quite figure out the right magic to make sdcc generate an adb file - I think I caught it whilst things were changing and never revisited it.

feilipu, 2026-11-08:
There has been a lot of work on -debug recently. If this is still current, please reopen.
@jorgegv has provided an updated interface.

closed 2026-11-08 by feilipu