z88dk/z88dk #2291: [zsdcc only][bug] extern definition for 'foo' mismatches with declaration previously defined

Fabrizio-Caruso, 2023-06-04:
`zsdcc` does not compile the simple code below (which is compiled with no issue by `sccz80` and even `gcc` with no warnings):
```
#include "extern.h"
extern const int foo[];
int main(void)
{
        return 0;
}
```
with `extern.h` only containing:
```
const int foo[] = {42,43,44,45,46,47};
```

You get this error:
```
$ zcc.exe +zx -compiler=sdcc extern.c
extern.c:3: error 91: extern definition for 'foo' mismatches with declaration.
extern.h:2: error 177: previously defined here
```
[extern.zip](https://github.com/z88dk/z88dk/files/11646614/extern.zip)

zx70, 2023-06-07:
I think the problem is in the name chosen for  your program.
extern is a reserved word and it is probably colliding with the exported module, named "extern".

Fabrizio-Caruso, 2023-06-07:
@zx70 here you are the same code with a different name. I was getting this error in Cross-Lib and I had to revert to sccz80 in all cases.
[intern.zip](https://github.com/z88dk/z88dk/files/11681620/intern.zip)

suborb, 2023-06-07:
Reroduced and reported upstream: https://sourceforge.net/p/sdcc/bugs/3597/

suborb, 2023-09-04:
This was fixed upstream in [r14247]

feilipu, 2023-09-04:
Its a small fix change. Probably worth adding to our 4.3.0 patch.

Will do it shortly.

closed 2023-09-05 by feilipu