z88dk/z88dk #2296: Library name selection

suborb, 2023-06-07:
For classic we're building some of the libraries (for examples maths) multiple times for different CPUs. Typically they should have names: `math_8080.lib math_8085.lib math.lib math_z180.lib math_z80n.lib math_ixiy.lib math_ez80.lib` etc

Picking the right library to use is at present left to the user (either explicitly or where we've added them via aliases) which is prone to mistakes.

Thinking about ways forward for this, I can see two ways:

1. Libraries are placed into lib/clibs/[cpu] and the appropriate path added as `-L`
2. z80asm at link time looked for a library with the pattern [name]_[cpu].lib (possibly with a fallback, for example ez80 -> z180 ->z80 -> no suffix)

I was originally going for 1. but I having doubts whether it's actually the best option. So thoughts...

feilipu, 2023-06-08:
Just some notes for reference, and to add to the "confusion".

- Related to #1568, which has a [partial solution branch](https://github.com/z88dk/z88dk/commit/ffa3d72048beeab4c3ce39b5ecfe504e22e6a370), which may or may not help. The rc2014 target in newlib has multiple hardware options for serial and disk access, and therefore needs to select the "driver" routines at build time via sub-types. This extends to the rc2014-8085, which is then a different machine, but with the same serial and disk access. Fortunately the matrix of alternative build options is currently very sparse.

- Unlike classic, newlib doesn't support link time selection of the integer math routines (fast or small or blended). This and many other options need to be [configured](https://github.com/z88dk/z88dk/wiki/Configuration) and then the target library rebuilt. Because there are so many options, it probably doesn't make sense to build every alternative configuration in advance. Perhaps that method would work for classic targets too?

- [ ] There are [integer math routines](https://github.com/z88dk/z88dk/blob/master/libsrc/_DEVELOPMENT/math/math_integer_z180.lst) for both `z180` and `zxn`, and they could also be used by `ez80` targets (argon). I don't think classic picks up this currently, because there wern't any `z180` machines targeted previously

- Both `math32` and `math16` are built for `z80`, `z180` and `zxn` machines, and newlib targets choose the appropriate machine library based on (I can't remember, but will check back when I figure out how this works).

- [ ] If anyone was interested, it would be possible (and relatively easy) to write a `r2ka` or `r3k` optimised set of multiply routines for `math32` or `math16` (which would be done via integer math support). There's no newlib Rabbit target today, so this wasn't done previously.

- `rc2014-8085` and `am9511_8085` are built for `8085` which is an odd one out for newlib, which doesn't usually have `8085` targets.

- And the big one. newlib has many math libraries built as targets `am9511`, `math16`, `math32`, and `m48`, with multiple machine options, and these libraries really shouldn't be targets.

suborb, 2023-06-09:
> Just a thought: the object file could include the machine for which it was assembled, and the library file could contain one version of the object file per machine, and the linker could select the right one.

I feel like the object file should contain the CPU regardless since the different CPUs aren't binary compatible.

Multi-arch libraries is a nice idea though I'm acutely aware that you've got the mountain of ez80 support underway.

pauloscustodio, 2023-06-09:
> Multi-arch libraries is a nice idea though I'm acutely aware that you've got the mountain of ez80 support underway.

Just finished! Waiting for CI to complete to open Pull Request.

I noticed you created ez80 and ez80_z80; in the assembler I have only ez80 and you need to enter in the source file ".assume adl=0/1" to select the mode.

suborb, 2023-06-09:
I think that's how the Zilog assembler works (I presume the default is ADL=0?). However, we'll need to detect the mode when assembling the library files (eg ez80 in ADL mode, `pop af` sets the top byte is 0 which will break a lot of existing code where we're juggling the stack).

I spotted that sdcc uses ez80_z80 to indicate 16 bit mode, so to reduce confusion I tried to follow it.

pauloscustodio, 2023-06-09:
> I can adapt to ez80/ez80_z80 and create a corresponding define for IFDEFs.

Done.

suborb, 2023-06-10:
I've plumped for option 2 with zcc doing some string replacement for the time being.

It's solved my main pain which is multiple aliases for maths libraries.

closed 2023-06-13 by suborb