z88dk/z88dk #2304: sccz80 am9511 8085 - comparison parameter generation

feilipu, 2023-06-13:
The am9511 8085 [comparison generation function](https://github.com/z88dk/z88dk/blob/master/libsrc/_DEVELOPMENT/math/float/am9511/asm/8085/am32_compare.asm) mutates its parameters, rather than popping them before use (like [the math32 z80 callee version](https://github.com/z88dk/z88dk/blob/master/libsrc/_DEVELOPMENT/math/float/math32/z80/f32_fscompare.asm#L52)). 

Normally `sccz80` will generate input parameters on the stack before calling any of the `l_[comparison]` functions, rather than using the original variables. But in some cases it doesn't. I noticed this when fixing the [`fmax()`](https://github.com/z88dk/z88dk/blob/master/libsrc/_DEVELOPMENT/math/float/am9511/c/am9511_fmax.c) and [`fmin()`](https://github.com/z88dk/z88dk/blob/master/libsrc/_DEVELOPMENT/math/float/am9511/c/am9511_fmin.c) functions, where I had to generate intermediate variables to protect the input variables, otherwise they would be overwritten and a corrupted maximum or minimum was returned.

I can't find the point in `sccz80` where there is a decision to generate temporary parameters as input to a comparison, or to use the original variables.

So the question is how do we make certain that the input parameters to a `l_[comparison]` operation are always generated, and don't refer directly to the original variables?

- [ ] on me to fix `am9511` math to use a callee compare function, so it is similar to `math32`.

suborb, 2023-06-13:
All the parameters should be temporary. I think the original problem was because the assembler written fmax/fmin reused its parameters for calling to the comparison function.

However, I do know you had a problem with the C variants, which I should have investigated at the time but didn't.

feilipu, 2023-06-13:
> However, I do know you had a problem with the C variants, which I should have investigated at the time but didn't.

It might have been operator error somewhere too. But I'm pretty sure that the assembly generated before I added the temporary variables was wrong, because it returned the same stack locations as were passed into it (and were destroyed by the mutating compare function).

[Incorrect function](https://github.com/z88dk/z88dk/blob/4a3d1a3f5b5ae97b0744660d02c31f357266b351/libsrc/_DEVELOPMENT/math/float/am9511/c/8085/am9511_fmax.asm). Ignore the wrong calling.

```asm
; Function am9511_fmax flags 0x00000000 __stdc 
; double am9511_fmax(double x, double y)
; parameter 'double x' at 2 size(4)
; parameter 'double y' at 6 size(4)
	C_LINE	13,"am9511_fmax.c::am9511_fmax::0::1"
._am9511_fmax
	ld	de,sp+2
	ex	de,hl
	call	l_glong2sp
	ld	de,sp+10
	ex	de,hl
	call	l_glong
	call	l_f32_gt
	ld	a,h
	or	l
	jp	z,i_2	;
	ld	de,sp+2
	ex	de,hl
	call	l_glong
	ret

.i_2
	ld	de,sp+6
	ex	de,hl
	call	l_glong
	ret
```

[Correct function](https://github.com/z88dk/z88dk/blob/master/libsrc/_DEVELOPMENT/math/float/am9511/c/8085/am9511_fmax.asm).

```asm
; Function am9511_fmax_callee flags 0x00000240 __smallc __z88dk_callee 
; double am9511_fmax_callee(double x, double y)
; parameter 'double y' at sp+2 size(4)
; parameter 'double x' at sp+6 size(4)
	C_LINE	13,"am9511_fmax.c::am9511_fmax_callee::0::0"
._am9511_fmax_callee
	ld	de,sp+6
	ex	de,hl
	call	l_glong2sp
	ld	de,sp+6
	ex	de,hl
	call	l_glong2sp
	ld	de,sp+4
	ex	de,hl
	call	l_glong2sp
	ld	de,sp+4
	ex	de,hl
	call	l_glong
	call	l_f32_gt
	ld	a,h
	or	l
	jp	z,i_2	;
	ld	de,sp+14
	ex	de,hl
	call	l_glong
	pop	bc
	pop	bc
	pop	bc
	pop	bc
	pop	bc
	pop	af
	pop	af
	pop	af
	pop	af
	push	bc
	ret

.i_2
	ld	de,sp+10
	ex	de,hl
	call	l_glong
	pop	bc
	pop	bc
	pop	bc
	pop	bc
	pop	bc
	pop	af
	pop	af
	pop	af
	pop	af
	push	bc
	ret
```

suborb, 2023-06-13:
I've put a couple of annotations on - it looks like it should be correct.
 
> ```assembly
> ; Function am9511_fmax flags 0x00000000 __stdc 
> ; double am9511_fmax(double x, double y)
> ; parameter 'double x' at 2 size(4)
> ; parameter 'double y' at 6 size(4)
> 	C_LINE	13,"am9511_fmax.c::am9511_fmax::0::1"
> ._am9511_fmax
> 	ld	de,sp+2
> 	ex	de,hl
> 	call	l_glong2sp. ; loads a long from sp+2 and pushes to stack
> 	ld	de,sp+10
> 	ex	de,hl
> 	call	l_glong.      ;loads a long into dehl
> 	call	l_f32_gt.    ;dumps 2 words on the stack
> 	ld	a,h
> 	or	l
> 	jp	z,i_2	;
> 	ld	de,sp+2
> 	ex	de,hl
> 	call	l_glong
> 	ret
> 
> 
> .i_2
> 	ld	de,sp+6
> 	ex	de,hl
> 	call	l_glong
> 	ret
> ```

closed 2023-06-13 by feilipu