z88dk/z88dk #2309: (z80asm) Request for emulated instructions

suborb, 2023-06-15:
The ez80 supports:

```
ld hl,(ix+d)
ld de,(ix+d)
ld b,(ix+d)
```

And the Rabbit supports:

```
ld hl,(ix+d)
```

It would be nice if z80asm could emulate the ez80 instructions on Rabbit, z80, z180 cpus

pauloscustodio, 2023-06-17:
Add emulation for:
ld (ix/id+d), bc/de/hl/ix/iy
ld bc/de/hl/ix/iy, (ix/iy+d)

pauloscustodio, 2023-06-17:
List of all opcodes recognized by z80asm and for which CPUs at:
https://raw.githubusercontent.com/z88dk/z88dk/master/src/z80asm/dev/cpu/opcodes.txt

suborb, 2023-06-17:
That opcode list is really useful - thank you. What's the command to regenerate it?

There's some that don't make sense (gbz80 doesn't have index registers):

```
|ld (hl), ix            |gbz80   |D5 DD E5 D1 73 23 72 D1 2B         |
|ld (hl+), ix           |gbz80   |D5 DD E5 D1 73 23 72 D1 23         |
|ld ix, %m              |gbz80   |DD 21 %m %m                        |
|ld ix, (hl)            |gbz80   |D5 DD E5 D1 5E 23 56 D1 2B         |
|ld ix, (hl+)           |gbz80   |D5 DD E5 D1 5E 23 56 D1 23         |
|ldi (hl), ix           |gbz80   |D5 DD E5 D1 73 23 72 D1 23         |
|ldi ix, (hl)           |gbz80   |D5 DD E5 D1 5E 23 56 D1 23         |
```

pauloscustodio, 2023-06-18:
@suborb , thank you for spotting this, fixed.

> Also, why XTHL should be accepted on Rabbits?

It is not, where did you see this?
```|xthl                   |8080 8085 ez80 ez80_z80 -     -    -   z180 z80 z80n|```

suborb, 2023-06-18:
> > Also, why XTHL should be accepted on Rabbits?
> 
> It is not, where did you see this? `|xthl |8080 8085 ez80 ez80_z80 - - - z180 z80 z80n|`

Sorry that’s bad wording by me. Why isn’t it accepted since the instruction does exist as ed 54?

commit d70311d7 (2023-06-17): Fix #2309 Request for emulated instructions

Add emulation for:
ld (ix/id+d), bc/de/hl/ix/iy
ld bc/de/hl/ix/iy, (ix/iy+d)
commit 914ad4dc (2023-06-17): Fix #2309 Request for emulated instructions

Add emulation for:
ld (ix/id+d), bc/de/hl/ix/iy
ld bc/de/hl/ix/iy, (ix/iy+d)
commit 82221b93 (2023-06-18): Fix #2309 Request for emulated instructions

Remove ix/iy instructions from 8080, 8085 and gbz80
commit f05d66e6 (2023-06-18): Fix #2309 Request for emulated instructions

Add xthl to Rabbits and GBZ80
closed 2023-06-18 by pauloscustodio