z88dk/z88dk #2312: (nightly) z88dk-dis is not available in bin of the nightly build

pauloscustodio, 2023-06-18:
But make install in ticks copies it to bin

suborb, 2023-06-18:
Windows I presume?

```
unzip -Z z88dk-win32-latest.zip | grep z88dk-dis
drwxrwxr-x  3.0 unx        0 bx stor 22-Jun-26 03:00 z88dk/win32/z88dk-dis/
-rw-rw-r--  3.0 unx     2104 tx defX 22-Jun-18 03:00 z88dk/win32/z88dk-dis/z88dk-dis.vcxproj.filters
-rw-rw-r--  3.0 unx     8849 tx defX 22-Jun-26 03:00 z88dk/win32/z88dk-dis/z88dk-dis.vcxproj
-rwxr-xr-x  3.0 unx   787231 bx defX 23-Jun-18 03:10 z88dk/bin/z88dk-dis.exe
-rwxr-xr-x  3.0 unx   673113 bx defX 23-Jun-18 03:11 z88dk/bin.x86/z88dk-dis.exe
```

pauloscustodio, 2023-06-18:
Sorry, my mistake; maybe extracted an old z88dk-win32-*.zip by mistake.

closed 2023-06-18 by pauloscustodio