z88dk/z88dk #2313: (dis) Incorrect disassembly for ez80_z80

pauloscustodio, 2023-06-18:
The following code for ez80_z80:
```
     1  0000  dd3708                    ld ix, (ix+8)
     2  0003  dd3f08                    ld (ix+8), ix
```
is disassembled as:
```
                    ld      iy,(ix+$08)                     ;[0000] dd 37 08
                    ld      (ix+$08),iy                     ;[0003] dd 3f 08
```

pauloscustodio, 2023-06-18:
This is closed. Thanks.

pauloscustodio, 2023-07-02:
Another case popped up, with ez80_z80:
```
     1  0000  ed31              ld iy,(hl)
     2
                    out0      ($00),(hl)                    ;[0000] ed 31 00
```

suborb, 2023-07-02:
Whilst on that opcode, this looks wrong:

```
dom@ermintrude ticks % cat ez80.asm

	ld	iy,(hl)
dom@ermintrude ticks % z88dk-z80asm -b ez80
dom@ermintrude ticks % z88dk-dis ez80.bin
                    push      de                            ;[0000] d5
                    push      iy                            ;[0001] fd e5
                    pop       de                            ;[0003] d1
                    ld        e,(hl)                        ;[0004] 5e
                    inc       hl                            ;[0005] 23
                    ld        d,(hl)                        ;[0006] 56
                    pop       de                            ;[0007] d1
                    dec       hl                            ;[0008] 2b
```

pauloscustodio, 2023-07-02:
Yes, this is wrong. Thanks

pauloscustodio, 2023-07-02:
Opened #2337

closed 2023-07-02 by pauloscustodio