z88dk/z88dk #2314: (dis) incorrect disassembly for gbz80

pauloscustodio, 2023-06-18:
On the gbz80, the ```stop``` instruction is ``10 00``. z88dk-dis outputs:
```
                    stop                                    ;[08ba] 10
                    nop                                     ;[08bb] 00
```

suborb, 2023-06-18:
I think there's some debate about this one, eg https://www.pastraiser.com/cpu/gameboy/gameboy_opcodes.html

> Instruction STOP has according to manuals opcode 10 00 and thus is 2 bytes long. Anyhow it seems there is no reason for it so some assemblers code it simply as one byte instruction 10.

So, I've handled it so that 0x10 0x00 is `stop` and 0x10 0xnn is a one byte `stop`

pauloscustodio, 2023-06-22:
In the gbz80 instruction ```add sp,N```, N is a signed integer.

The code
```
     1  0000  e880              add sp,-$80
```
is disassembled as:
```
                    add       sp,$80                        ;[0000] e8 80
```

and outputs a warning when assembled again:
```
test.asm:1: warning: integer range: $80
  ^---- 128
```

pauloscustodio, 2023-06-22:
Same problem:
```
ld hl,sp-128

```

pauloscustodio, 2023-06-22:
Closed by mistake. Sorry.

pauloscustodio, 2023-06-22:
Closed. Thanks.

closed 2023-06-22 by pauloscustodio