z88dk/z88dk #2315: (dis) incorrect disassembly for Rabbit

pauloscustodio, 2023-06-18:
The ALTD, IOE and IOI prefixes should appear as prefix to the following instruction, and not as single instructions.

This allows also the output of the disassembler to be fed back to the assembler.

suborb, 2023-06-18:
It messes up the formatting a little, but I've done this.

Related: I'm not sure of the best way to deal with it (regarding feeding back into the assembler), but the ez80 disassembler uses the full suffixes (sis, sil, lis, lil) rather than the short forms.

pauloscustodio, 2023-06-20:
After a long detour because of the incorrect assembly of "altd djnz asmpc", there's this other error in z88dk-dis for the Rabbit:

```
  5833  57c2  dd650080           ldp (-32768), ix               ; DD 65 00 80
                    ldp       ($8000),hl                    ;[57c2] dd 65 00 80
```

From the Rabbit manual:
ED 65 n m LDP (mn),HL 
DD 65 n m LDP (mn),IX 
FD 65 n m LDP (mn),IY

pauloscustodio, 2023-06-21:
Another one:
```
     1  0000  5b                idet
     2  0001  765b              ld e',e
                    idet                                    ;[0000] 5b
                    altd idet                               ;[0001] 76 5b
```
From the manual: "If IDET is prefixed by ALTD, the opcode LD E’,E is executed"

pauloscustodio, 2023-06-21:
@suborb, "idet" is the last of the misses I found with the round-robin assemble all opcodes - disassemble - assemble again - compare binaries.

Thanks for your fixes.

suborb, 2023-06-21:
@pauloscustodio no worries, thanks for finding them!