z88dk/z88dk #2331: Does not build for ARM hosts

AshleighTheCutie, 2023-06-29:
As the title says. Everything but `z88dk-z80asm` builds, making the entire software suite useless. The only modification I made to the `./build.sh` file is to have `make` use all 6 cores of the devboard (Odroid N2+ 2GB). It does this with all source code download forms, regardless of if it is downloaded with the `git clone --recursive` for the latest version, or the source tarball for v2.2.

```
ashleigh@SmolLinuxBox:~/z88dk$ ./build.sh
make -C src/appmake PREFIX=`pwd` install
make -C src/copt PREFIX=`pwd` install
make -C src/cpp PREFIX=`pwd` install
make -C src/ucpp PREFIX=`pwd` install
make -C src/sccz80 PREFIX=`pwd` install
make -C src/z80asm PREFIX=`pwd` PREFIX_SHARE=`pwd` install
make[1]: Entering directory '/home/ashleigh/z88dk/src/copt'
ccache gcc -c -o copt.o -g -O2 -std=gnu99 -Wall -pedantic -DLOCAL_REGEXP -I. copt.c
make[1]: Entering directory '/home/ashleigh/z88dk/src/ucpp'
ccache gcc -g -O2 -Wall -std=gnu11 -MMD -I../../ext/optparse -I../../ext/regex -DSTAND_ALONE -DUCPP_CONFIG   -c -o mem.o mem.c
make[1]: Entering directory '/home/ashleigh/z88dk/src/cpp'
ccache gcc -g -O2 -Wall -std=gnu11 -MMD -I../../ext/optparse -I../../ext/regex    -c -o cpp1.o cpp1.c
make[1]: Entering directory '/home/ashleigh/z88dk/src/z80asm'
make[1]: Entering directory '/home/ashleigh/z88dk/src/sccz80'
ccache gcc -g -O2 -MMD -Wall -I../common -I../../ext/uthash/src/ -g -pedantic -std=gnu99   -c -o callfunc.o callfunc.c
make[1]: Entering directory '/home/ashleigh/z88dk/src/appmake'
ccache gcc -g -O2 -Wall -std=gnu11 -MMD -pedantic -I../common   -c -o abc80.o abc80.c
ccache gcc -g -O2 -Wall -std=gnu11 -MMD -I../../ext/optparse -I../../ext/regex -DSTAND_ALONE -DUCPP_CONFIG   -c -o nhash.o nhash.c
ccache gcc -std=gnu11 -MMD -Wall -O3 -Isrc -Isrc/c -It -I../common -I../../ext/optparse -I../../ext/regex -I../../ext/uthash/src  -g -O2 -c -o src/c/alloc.o src/c/alloc.c
abc80.c: In function ‘datablockout’:
abc80.c:129:17: warning: ignoring return value of ‘fread’ declared with attribute ‘warn_unused_result’ [-Wunused-result]
  129 |                 fread(&buffer[3], 253, 1, fin);
      |                 ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
ccache gcc -g -O2 -Wall -std=gnu11 -MMD -pedantic -I../common   -c -o ace-tap.o ace-tap.c
ccache gcc -g -O2 -Wall -std=gnu11 -MMD -I../../ext/optparse -I../../ext/regex -DSTAND_ALONE -DUCPP_CONFIG   -c -o cpp.o cpp.c
ccache gcc -g -O2 -Wall -std=gnu11 -MMD -pedantic -I../common   -c -o adam.o adam.c
ccache gcc -g -O2 -Wall -std=gnu11 -MMD -pedantic -I../common   -c -o appmake.o appmake.c
appmake.c: In function ‘option_set’:
appmake.c:576:34: warning: ISO C forbids conversion of object pointer to function pointer type [-Wpedantic]
  576 |                 option_cb func = (option_cb)option->dest;
      |                                  ^
appmake.c: In function ‘mb_enumerate_banks’:
appmake.c:1223:65: warning: ‘_’ directive output may be truncated writing 1 byte into a region of size between 0 and 4096 [-Wformat-truncation=]
 1223 |                             snprintf(bfilename, MBBUFFERMAX, "%s_%s.bin", binname_buf, section_name);
      |                                                                 ^
In file included from /usr/include/stdio.h:894,
                 from appmake.h:11,
                 from appmake.c:14:
/usr/include/aarch64-linux-gnu/bits/stdio2.h:71:10: note: ‘__builtin___snprintf_chk’ output between 6 and 5120 bytes into a destination of size 4096
   71 |   return __builtin___snprintf_chk (__s, __n, __USE_FORTIFY_LEVEL - 1,
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
   72 |                                    __glibc_objsize (__s), __fmt,
      |                                    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
   73 |                                    __va_arg_pack ());
      |                                    ~~~~~~~~~~~~~~~~~
ccache gcc -g -O2 -Wall -std=gnu11 -MMD -pedantic -I../common   -c -o aquarius.o aquarius.c
ccache gcc -g -O2 -Wall -std=gnu11 -MMD -pedantic -I../common   -c -o c128.o c128.c
ccache gcc -g -O2 -Wall -std=gnu11 -MMD -pedantic -I../common   -c -o c7420.o c7420.c
ccache gcc -g -O2 -Wall -std=gnu11 -MMD -I../../ext/optparse -I../../ext/regex    -c -o cpp2.o cpp2.c
ccache gcc -std=gnu11 -MMD -Wall -O3 -Isrc -Isrc/c -It -I../common -I../../ext/optparse -I../../ext/regex -I../../ext/uthash/src  -g -O2 -c -o src/c/array.o src/c/array.c
ccache gcc -g -O2 -Wall -std=gnu11 -MMD -I../../ext/optparse -I../../ext/regex    -c -o cpp3.o cpp3.c
ccache gcc -g -O2 -Wall -std=gnu11 -MMD -pedantic -I../common   -c -o cpc.o cpc.c
ccache gcc -c -o regex/regcomp.o -g -O2 -std=gnu99 -Wall -pedantic -DLOCAL_REGEXP -I. regex/regcomp.c
ccache gcc -c -o regex/regerror.o -g -O2 -std=gnu99 -Wall -pedantic -DLOCAL_REGEXP -I. regex/regerror.c
ccache gcc -c -o regex/regexec.o -g -O2 -std=gnu99 -Wall -pedantic -DLOCAL_REGEXP -I. regex/regexec.c
ccache gcc -c -o regex/regfree.o -g -O2 -std=gnu99 -Wall -pedantic -DLOCAL_REGEXP -I. regex/regfree.c
ccache gcc -g -O2 -MMD -Wall -I../common -I../../ext/uthash/src/ -g -pedantic -std=gnu99   -c -o cdbfile.o cdbfile.c
ccache gcc -o z88dk-copt  copt.o regex/regcomp.o  regex/regerror.o regex/regexec.o  regex/regfree.o
install z88dk-copt /home/ashleigh/z88dk/bin/z88dk-copt
make[1]: Leaving directory '/home/ashleigh/z88dk/src/copt'
ccache gcc -g -O2 -Wall -std=gnu11 -MMD -pedantic -I../common   -c -o cpm2.o cpm2.c
ccache gcc -g -O2 -Wall -std=gnu11 -MMD -I../../ext/optparse -I../../ext/regex    -c -o cpp4.o cpp4.c
ccache gcc -std=gnu11 -MMD -Wall -O3 -Isrc -Isrc/c -It -I../common -I../../ext/optparse -I../../ext/regex -I../../ext/uthash/src  -g -O2 -c -o src/c/class.o src/c/class.c
ccache gcc -g -O2 -Wall -std=gnu11 -MMD -I../../ext/optparse -I../../ext/regex    -c -o cpp5.o cpp5.c
cpm2.c:40:95: warning: ISO C forbids conversion of function pointer to object pointer type [-Wpedantic]
   40 |     { 'a', "add-file", "Add additional files [hostfile:cpmfile] or [hostfile]", OPT_FUNCTION, (void *)c_add_file },
[... 589 lines cut ...]
ccache gcc -g -O2 -Wall -std=gnu11 -MMD -pedantic -I../common -o z88dk-appmake abc80.o ace-tap.o adam.o appmake.o aquarius.o c128.o c7420.o cpc.o cpm2.o cpmdisk.o dai.o enterprise.o extract.o fat.o ff.o fp1100.o galaksija.o gb.o glue.o hex.o homelab.o inject.o kc.o lviv.o lynx.o lz49.o m5.o mamepp.o mameql.o mc.o mgt.o msx.o msxrom.o mtx.o multi8.o mz2500.o mz.o nascom.o nec.o newbrain.o newext.o noop.o ondra.o p2000.o pasopia7.o pc88.o pc88disc.o pmd85.o primo.o px.o residos.o rex6000.o rk.o rom.o sc3000.o smc777.o sms.o sol20.o sorcerer.o sos.o spc1000.o svi.o tixx.o trs80.o tvc.o vector06c.o vg5k.o vz.o x07.o x1.o z1013.o z88.o z88elf.o z88shell.o z9001.o zx81.o zx.o zxn.o zx-util.o zxvgs.o ../common/dirname.o 
install z88dk-appmake /home/ashleigh/z88dk/bin/z88dk-appmake
make[1]: Leaving directory '/home/ashleigh/z88dk/src/appmake'
ccache gcc -g -O2 -I../../ext/uthash/src/ -g -MMD    -c -o acia.o acia.c
ccache gcc -g -O2 -I../../ext/uthash/src/ -g -MMD    -c -o hook_rc2014.o hook_rc2014.c
ccache gcc -g -O2 -I../../ext/uthash/src/ -g -MMD    -c -o debug.o debug.c
ccache gcc -g -O2 -I../../ext/uthash/src/ -g -MMD    -c -o srcfile.o srcfile.c
ccache gcc -g -O2 -I../../ext/uthash/src/ -g -MMD    -c -o lex.yy.o lex.yy.c
ccache gcc -g -O2 -I../../ext/uthash/src/ -g -MMD    -c -o expressions.tab.o expressions.tab.c
ccache gcc -g -O2 -I../../ext/uthash/src/ -g -MMD    -c -o disassembler_main.o disassembler_main.c
ccache gcc -g -O2 -I../../ext/uthash/src/ -g -MMD    -c -o debugger_gdb.o debugger_gdb.c
ccache gcc -g -O2 -I../../ext/uthash/src/ -g -MMD    -c -o debugger_mi2.o debugger_mi2.c
ccache gcc -g -O2 -I../../ext/uthash/src/ -g -MMD    -c -o debugger_gdb_packets.o debugger_gdb_packets.c
ccache gcc -g -O2 -I../../ext/uthash/src/ -g -MMD    -c -o sxmlc.o sxmlc.c
ccache gcc -g -O2 -I../../ext/uthash/src/ -g -MMD    -c -o sxmlsearch.o sxmlsearch.c
ccache gcc -o z88dk-dis -g -O2 -I../../ext/uthash/src/ -g -MMD  disassembler_main.o syms.o disassembler_alg.o debug.o exp_engine.o backend.o
ccache gcc -o z88dk-gdb -g -O2 -I../../ext/uthash/src/ -g -MMD  cpu.o backend.o syms.o disassembler_alg.o debug.o exp_engine.o debugger.o breakpoints.o profiler.o debugger_gdb.o debugger_mi2.o debugger_gdb_packets.o linenoise.o srcfile.o ../common/dirname.o sxmlc.o sxmlsearch.o  lex.yy.o expressions.tab.o -lm -lpthread
make[1]: Leaving directory '/home/ashleigh/z88dk/src/z80asm'
make: *** [Makefile:142: bin/z88dk-z80asm] Error 2
[... cut ...]
ccache gcc -o z88dk-ticks -g -O2 -I../../ext/uthash/src/ -g -MMD  ticks.o cpu.o backend.o hook_cpm.o hook_console.o hook_io.o hook_misc.o hook.o debugger.o breakpoints.o profiler.o exp_engine.o debugger_ticks.o linenoise.o utf8.o syms.o disassembler_alg.o memory.o am9511.o acia.o hook_rc2014.o debug.o srcfile.o ../common/dirname.o  lex.yy.o expressions.tab.o -lm
install z88dk-ticks /home/ashleigh/z88dk/bin/z88dk-ticks
install z88dk-dis /home/ashleigh/z88dk/bin/z88dk-dis
install z88dk-gdb /home/ashleigh/z88dk/bin/z88dk-gdb
make[1]: Leaving directory '/home/ashleigh/z88dk/src/ticks'```

ImperatorStorm, 2023-06-29:
Can reproduce in a clean Ubuntu arm64 vm.

feilipu, 2023-06-29:
Pretty sure that multi-core is not fully complete yet.

Try leaving the `.build.sh` alone and building with single core , and report.

suborb, 2023-10-22:
Superseded by #2419

closed 2023-10-22 by suborb