z88dk/z88dk #2333: Binary content in a defm statement generates unexpected results

suborb, 2023-06-29:
This has come out of the clang work:

Given this (3 byte string 0x01, 0xx18, 0xff)

```
_A_GD_FLIN:
        defm    "^A^X<FF>"
```

```
hexdump -C x.asm
00000000  5f 41 5f 47 44 5f 46 4c  49 4e 3a 0a 09 64 65 66  |_A_GD_FLIN:..def|
00000010  6d 09 22 01 18 ff 22 0a                           |m."...".|
00000018
```

z88dk-z80asm -b [file] generates:

```
0000000 0001 2d18 0031
0000005
```

pauloscustodio, 2023-06-30:
@suborb , thanks. This one is very old!

suborb, 2023-07-01:
Thanks, I've now got a related one. Overlong line perhaps?

[chessboard.c.txt](https://github.com/z88dk/z88dk/files/11927232/chessboard.c.txt)

pauloscustodio, 2023-07-02:
This one has a linefeed inside the quotes:

```
000001a0                                                         22 20 40 3e 20  |       " @> |
000001b0  20 21 c0 20 10 22                                                    | !. .", 0, "..",|
```

As the assembler reads line by line, it splits the line at the LF char.

The text would have to contain "\x10" instead.

pauloscustodio, 2023-07-02:
Is this a bug? should we allow multi-line quoted strings?

suborb, 2023-07-02:
I‘m not sure!

I don’t think I can fix it in either clang or workaround it in copt. 

I’m also not sure how much would break of it was allowed by default. 

We could support it under a different directive?

pauloscustodio, 2023-07-02:
Opened #2338 and closing this one.

commit e7ebdc85 (2023-06-30): Fix #2333 Binary content in a defm statement

generates unexpected results
closed 2023-07-02 by pauloscustodio