z88dk/z88dk #2340: (dis) Incorrect disassembly for ez80_z80

pauloscustodio, 2023-07-02:
With ez80_z80, the code:
```
  2221  1aa2  5b320080ff         ld.il (-32768), a              ; 5B 32 00 80 FF
```
disassembles as:
```
                    ld.lil    ($ff8000),a                   ;[1aa2] 5b 32 00 80 ff
```
But the equivalent long suffix of .il is .sil in adl=0 and .lil in adl=1 (page 24 of ez80_cpu_um77.pdf)

This should be ld.sil.

suborb, 2023-07-02:
Is that not an assembler issue? With a `.il` suffix, the assembler should be inserting `.sil` so 0x52? 

[image]

pauloscustodio, 2023-07-03:
Correct, this was a bug in z80asm.

Now one in dis ez80_z80:
```49 [09|19|29|30]``` should disassemble ```add.l hl, rr```, it disassembles ```add hl, rr```; and the same for ```49 [DD|FD] [09|19|29|30]```.

pauloscustodio, 2023-07-03:
Another one: ```49 35``` disassembles ```dec (hl)```, should be ```dec.l (hl)```

pauloscustodio, 2023-07-03:
Another one: ```49 34``` disassembles ```inc (hl)```, should be ```inc.l (hl)```

pauloscustodio, 2023-07-03:
A strange one, the disassembly is wrong only on ez80:
```
     1  0000  dd360304          ld (ix+3), 4
```
z88dk-dis test.bin:
```
                    ld        (ix+$03),$04                  ;[0000] dd 36 03 04
```
z88dk-dis -mez80_z80 test.bin:
```
                    ld        (ix+$03),iy                   ;[0000] dd 36 03
                    inc       b                             ;[0003] 04
```

pauloscustodio, 2023-07-03:
```
  2483  1e73  49ed31             ld.l iy, (hl)                  ; 49 ED 31
```
disassembles:
```
  2942  1e73  ed31                                  ld        iy,(hl)                       ;[1e73] 49 ed 31
```

pauloscustodio, 2023-07-03:
```
  4519  3828  49ed34             tst.l (hl)                    ; 49 ED 34
```
disassembles:
```
  5407  3828  ed34                                  tst       (hl)                          ;[3828] 49 ed 34
```

suborb, 2023-07-03:
Thanks, those two are dealt with - can't be many more left!

pauloscustodio, 2023-07-04:
I don't know; each time it fails I have to dig into a hex-dump to find the difference, and only one failure at a time. But it is fun.

pauloscustodio, 2023-07-04:
```
  2483  1e73  49ed31             ld.l iy, (hl)                  ; 49 ED 31
```
disassembles as:
```
  2942  1e73  ed31                                  ld        iy,(hl)                       ;[1e73] 49 ed 31
```

pauloscustodio, 2023-07-04:
Thanks. All done.

Now we could think about supporting ez80 adl mode, if you have interest. From my side I can just flip the switch to test it also.

suborb, 2023-07-04:
The disassembler should support -mez80 to default to ADL=1 mode.

pauloscustodio, 2023-07-04:
Now in -mez80 mode (ADL=1), the code:
```
     1  0000  49cd3412          call.lis 0x1234
```
disassembles as:
```
                    call.lis  $001234                       ;[0000] 49 cd 34 12 00
```

pauloscustodio, 2023-07-04:
Again with ADL=1, the code:
```
     1  0000  40c3ffff          jp.sis 65535
```
disassembles:
```
                    jp.sis    $00ffff                       ;[0000] 40 c3 ff ff 00
```

pauloscustodio, 2023-07-04:
With ADL=1, the code:
```
     1  0000  4932ffff          ld.is (65535), a
```
disassembles:
```
                    ld.lis    ($00ffff),a                   ;[0000] 49 32 ff ff 00
```

pauloscustodio, 2023-07-05:
All done. Thanks.

commit 3241b0e2 (2023-07-04): Fix #2340 (z80asm) fixed ld.il, ld.is, ret*.lil, ret*.lis
commit 85f4b63b (2023-07-05): Fix #2340 (z80asm) assemble-disassemble-assemble

for all supported CPUs
commit 40a1c416 (2023-07-05): Fix #2340 (z80asm) assemble-disassemble-assemble

for all supported CPUs
closed 2023-07-05 by pauloscustodio