z88dk/z88dk #2341: Unable to build z88dk

TurBoss, 2023-07-02:
Hello,

Ive been following git commits and just found that I cannot build any more, this is the latest working commit for me : d187ba665a08b50a34d7e17e0eaf3612a2b43f4a

this is the error i get:

```
zcc +test -vn -a Issue_1466_float16_callee.c -o Issue_1466_float16_callee.opt.tmp1
diff -w Issue_1466_float16_callee.opt.tmp2 results/Issue_1466_float16_callee.opt
13a14,15
> 	GLOBAL	___powf16
> .___powf16
make[1]: *** [Makefile:49: Issue_1466_float16_callee.opt] Error 1
make[1]: se sale del directorio '/home/turboss/Projects/z88dk/testsuite'
make: *** [Makefile:227: testsuite] Error 2

```

Thanks

suborb, 2023-07-02:
I think you need to rebuild sccz80 - force a rebuild by removing bin/z88dk-sccz80

suborb, 2023-07-02:
I assume that you're sorted - reopen if not!

closed 2023-07-02 by suborb

TurBoss, 2023-07-02:
I cleaned bin/z88dk and updated submodules

its building again. 

Thank you!