z88dk/z88dk #2343: [feature/question] How can we set RAM_Start, RAM_Length, Stack_top ?

Fabrizio-Caruso, 2023-07-05:
How can we set RAM_Start, RAM_Length, Stack_top ?
CC65 does this through a linker configuration file. How is it done in Z88DK? 

Some targets may have RAM at different locations depending on their configuration or the developer may want to spare some RAM areas for other reasons. So if it is not already possible, it would be very useful.

This would also help me with the issues I have in #2342

suborb, 2023-07-05:
I think sc3000 has the old way of doing it, it needs an update to support CRT_ORG_BSS. The stack can be moved using` -pragma-define:REGISTER_SP=xxx`

Fabrizio-Caruso, 2023-07-05:
@suborb Thanks.
-pragma-define:REGISTER_SP=xxx may already allow me to use 2k on the SC3000 rom subtype.

As I explain the default value for Stack_top is probably wrong even for the rom subtype because RAM_Length is $0800 (2K) and the Stack is at 1K offset, so the ram above it is declared available but it cannot be used as the stack is in the middle of the ram.
Either both at 1K (as for the SG1000) or much better both at 2K (as on the SC3000) because mirroring should make it work on both machines.

So, for the non-rom standard case, we will need an update.
Moreover the default non-rom should most probably have RAM_Start at $8000.

suborb, 2023-07-05:
I've switched the default to the sc3000 (since that's the name of the target) so assumes its a 2k memory.

For the RAM types, I think it's designed to coexist with BASIC - the start address is easily changed with `-pragma-define:CRT_ORG_CODE=0x8000` for example.

suborb, 2023-07-05:
I should read the documentation I wrote here (https://github.com/z88dk/z88dk/wiki/Platform---Sega-SC3000):

> The ROM produced is compatible for the SG-1000 and SC-3000. This means that it assumes only 1k of memory and places | the stack accordingly. A SC-3000 only ROM can produced by moving the stack pointer with the option: -pragma-define:REGISTER_SP=0xc7f0

Fabrizio-Caruso, 2023-07-05:
@suborb I think -pragma-define:REGISTER_SP=0xc7f0 (why not -pragma-define:REGISTER_SP=0xc7ff ?)
should work on the sg1000 because c7xx is mirrored to c3xx on the SG1000.

For the non case, we are assuming BASIC and BASIC may even disable the work at $c000-$c7ff as for Level III B. So I am not sure it is good to have the default memory at $c000 instead of $8000, which should work straightaway on most BASIC cases. Anyway this is not import as long as -pragma-define:CRT_ORG_CODE=0x8000 can be used.

Fabrizio-Caruso, 2023-07-05:
I suppose you mean -pragma-define:CRT_ORG_BSS=0x8000

suborb, 2023-07-05:
I think I’m getting confused!

There’s a RAM target that is meant to load from tape into basic. We compile to start from $9317 or something like that. This is what I thought you were talking about when you were talking about BASIC. 

There’s a ROM target which by default runs on an unexpanded machine. 

I now think you’re talking about the running the ROM target on an expanded machine. In which case setting CRT_ORG_BSS and REGISTER_SP will be needed to take advantage of the extra ram.

suborb, 2023-07-05:
> I suppose you mean -pragma-define:CRT_ORG_BSS=0x8000
> Right?
> What about the other RAM segments? Like DATA and RODATA? Is CRT_ORG_BSS taking care of all of them?

Data is bundled into bss for org purposes on ROM targets. Rodata is always in ROM.

Fabrizio-Caruso, 2023-07-05:
Why do you compile at $9317? Is it necessary because of the BASIC loader?

Concerning the ROM target, I understand that the current set-up runs on both the SG1000 and SC3000 as long as you do not use more than 1k of RAM.

What I was proposing is to have default Stack_top for the ROM target at $C7FF instead of $C3FF. Because of mirroring, if I am not wrong, this should continue to work on SG1000 and SC3000 as long as you do not use more than 1k of RAM and on top of that it work on the SC3000 if you use up to 2k with no need to have an extra option. It is not really important now, though, because I see I can set the stack higher.

closed 2023-07-08 by Fabrizio-Caruso