z88dk/z88dk #2345: [bug] sleep hangs the computer on multiple targets

Fabrizio-Caruso, 2023-07-05:
sleep hangs the computer on multiple targets.

For example the following code will just print 0 and 1 but never reach 2
```
#include <stdlib.h>
#include <stdio.h>

int main(void)

{
	short i;
	
	for(i=0;i<10;++i)
	{
		printf("%d\n", i);
		sleep(1);
	}
	while(1){};
	return 0;
}
```

I am getting this problem on multiple targets, e.g., zx and msx.

For example:
```
zcc +zx ./debug/test_sleep.c -create-app -o test_sleep -lndos
```

suborb, 2023-07-05:
Sorry about that, should be good now.

closed 2023-07-05 by suborb

Fabrizio-Caruso, 2023-07-06:
@suborb There is no nightly build, though. The snapshots for both today and yesterday are missing. The build process might have failed.

commit 85f05348 (2023-07-05): Fix #2345: Clangification broke sleep