z88dk/z88dk #2349: [agon][bug] Wrong colors in mode 2

Fabrizio-Caruso, 2023-07-07:
In my games, some colors are mapped to the wrong color.
For example red is show as blue.

suborb, 2023-07-07:
It's known, see: https://z88dk.org/forum/viewtopic.php?p=21941#p21941

The emulator still doesn't support redefining colours so I probably ought to add a mapping in.

suborb, 2023-07-07:
I've applied the following mappings:

```
        defb    $0      ;BLACK -> (#00 00 00)
        defb    12      ;BLUE -> (#00 00 ff)
        defb    10      ;GREEN -> (#00 ff 00)
        defb    14      ;CYAN -> (#00 ff ff)
        defb    9       ;RED -> (#ff 00 00)
        defb    13      ;MAGENTA -> (#ff 00 ff)
        defb    43      ;BROWN -> (#aa 55 55)
        defb    15      ;LIGHTGRAY -> (#ff ff ff)
        defb    15      ;DARKGRAY -> (#ff ff ff)
        defb    47      ;LIGHTBLUE -> (#aa aa ff)
        defb    2       ;LIGHTGREEN -> (#00 aa 00)
        defb    51      ;LIGHTCYAN -> (#aa ff ff)
        defb    1       ;LIGHTRED -> (#aa 00 00)
        defb    57      ;LIGHTMAGENTA -> (#ff 55 ff)
        defb    11      ;YELLOW -> (#ff ff 00)
        defb    15      ;WHITE -> (#ff ff ff)
```

Available colours are listed here: https://gist.github.com/astralaster/33a7c4ca85380925b306b2df37afbd19 - corrections/alterations welcome.

Fabrizio-Caruso, 2023-07-08:
Thanks!
This is now fixed!
[image]

closed 2023-07-08 by Fabrizio-Caruso