z88dk/z88dk #2350: [agon][bug] Scroll-up triggered by displaying lowest-right character

Fabrizio-Caruso, 2023-07-07:
Scroll-up triggered by displaying lowest-right character.

Test in mode 2 with this emulator: https://github.com/astralaster/agon-light-emulator?fbclid=IwAR2Rg22XZCpnR36FQ80eeMRIA7TuNaxKh0f4jPRKAae2wHmeSXBscfXeRK8

suborb, 2023-07-07:
This is in the VDP (https://github.com/breakintoprogram/agon-vdp/blob/main/video/video.ino), the only thing I can do is to reduce the number of rows by 1.

```
        if(c >= 0x20 && c != 0x7F) {
                Canvas->drawChar(activeCursor->X, activeCursor->Y, c);
                cursorRight();
```

and 

```
void cursorRight() {
        activeCursor->X += fontW;
        if(activeCursor->X > activeViewport->X2) {
        cursorHome();
        cursorDown();
        }
}
```

Fabrizio-Caruso, 2023-07-08:
My current trivial work-around in Cross-Lib is to just avoid the lowest row. This works in my case.
So, is it really a limit of the hardware or is it a limit of the firmware or software they provide?

suborb, 2023-07-08:
It’s the way the firmware is written - it does the scroll when the character is printed rather than setting a flag and scrolling if it needs to when printing the next character. 

There might be a toggle to stop the behaviour - perhaps using the graphics cursor rather than the text cursor which I can look into. 

There’s a couple of other targets that have the same problem (I can’t remember which ones unfortunately).

Fabrizio-Caruso, 2023-07-08:
As far as the Agon is concerned, I have been told that they are working on a solution...

Fabrizio-Caruso, 2023-07-08:
This may be a fix for it:
https://github.com/breakintoprogram/agon-vdp/issues/55?fbclid=IwAR3ZtDaFNVG55XcxoeyxKfzYCAyROnNOCU59cR89C9il-VovwzMzZs_Dvto

closed 2023-07-08 by Fabrizio-Caruso

suborb, 2023-07-08:
Thanks, I've updated the wiki page with a link to the issues.