z88dk/z88dk #2351: [agon][bug?improvement] Keyboard input is laggy or even blocked

Fabrizio-Caruso, 2023-07-07:
While testing my games on the emulator I have experienced laggy keyboard input and even input blocked until a key is released and repressed.

Compiled with
```
agon_mode2: $(ASSETS_PATH)/z88dk_xchase.asm
	$(Z88DK_PATH)$(MYZ88DK) +agon $(SCCZ80_OPTS) $(INCLUDE_OPTS) \
	-D__AGON__ -D_XL_NO_JOYSTICK -pragma-redirect:fputc_cons=fputc_cons_generic \
	-D__SCREEN_MODE=2 -pragma-define:CRT_ENABLE_COMMANDLINE=0 \
	-D__CONIO_GRAPHICS -DXSize=40 -DYSize=24 \
	-D_XL_SLOW_DOWN_FACTOR=$(_COMMON_SLOWDOWN_FACTOR)*$(_AGON_SLOWDOWN) \
	$(CROSS_LIB_PATH)/display/init_graphics/z88dk/udg/udg_init_graphics.c \
	$(ASSETS_PATH)/z88dk_xchase.asm \
	$(FULL_FILES) \
	-o $(BUILD_PATH)/X$(GAME_NAME)_$@
```

Tested in mode 2 with this emulator: https://github.com/astralaster/agon-light-emulator?fbclid=IwAR2Rg22XZCpnR36FQ80eeMRIA7TuNaxKh0f4jPRKAae2wHmeSXBscfXeRK8

suborb, 2023-07-07:
Is it reproducible with this simple program (I'm wondering if it's an emulator/SDL issue):

```
#include <stdio.h>
#include <stdlib.h>

int main() {
        int c;

        printf("Going\n");
        while ( 1 ) {
                c = getk();
                printf(".");
                if ( c != 0 ) printf("Got key %d ", c);
                msleep(100);
        }

}
```

Fabrizio-Caruso, 2023-07-08:
I have been told that this is a bug on the current Agon.

Fabrizio-Caruso, 2023-07-08:
Do we close this issue? Or do we keep it as a reminder to recheck the behavior?

suborb, 2023-07-08:
If you've got a link to a thread elsewhere, I'm happy to update the wiki page with a bugs/issues section rather than keep this issue open. Likewise for the other issue.

Fabrizio-Caruso, 2023-07-08:
I was told this is a fix:
https://github.com/breakintoprogram/agon-mos/pull/73?fbclid=IwAR0-6BF3Ro_tuxaHoWTq4KfIJAnw18YvTvyhFAu9EMmcb5PJc8vK5y-TwZc

closed 2023-07-08 by Fabrizio-Caruso