z88dk/z88dk #2352: (classic) Add support for Aquarius+

suborb, 2023-07-08:
New z80 board with evolution of the Aquarius.

Tech specs here: https://github.com/fvdhoef/aquarius-plus/blob/master/docs/aquarius_plus.md

Emulators are available.

suborb, 2023-07-15:
With the font now being in ROM, makes that Spectrum look very easy (this is still text mode):

[image]

dbolli, 2024-10-08:
If anyone is interested, I've ported the [Aquarius+ tile and sprite demo](https://github.com/fvdhoef/aquarius-plus/tree/master/SDK/c/examples/tiles_and_sprites) to z88dk. Source code [here](https://www.dropbox.com/scl/fi/aie7bs76q9cex4myuna18/aquariusplus-tiles-and-sprites-example.zip?rlkey=2kwjrbw3ff4nkjbq6qrc19zgu&dl=1).

Great work supporting this platform that I was previously mostly unaware of 👍

suborb, 2024-10-08:
Ah, wonderful. If you don't mind I'll incorporate that!

It's a really nice platform, probably my favourite of the modern updates I've added support for (Agon, Next, Aq+), of course it's the one I don't have as hardware!

dbolli, 2024-10-08:
> Ah, wonderful. If you don't mind I'll incorporate that!

Please do 🙂

> It's a really nice platform, probably my favourite of the modern updates I've added support for (Agon, Next, Aq+), of course it's the one I don't have as hardware!

I agree. I'm having lots of fun experimenting with all the Aquarius+ PlusBASIC and asm features.