z88dk/z88dk #2353: [question/bug?][microbee] How to load .BEE files into an emulator?

Fabrizio-Caruso, 2023-07-09:
Hi!
I have not been able to load any of my games with Nanowasp (nanowasp.org).
I have tried to run UBEE512 but it clicking on any of the .bat files won't start the emulator (under Windows).

Which commands should I use to run .BEE files produced by Z88DK (appmake)?

suborb, 2023-07-09:
The instructions on the wiki page are a bit vague and not overly useful - I've really struggled with this target in the past (which is one of the reasons why the graphics drivers aren't unified).

I've tended to use CP/M on the Microbee. Having said that, ubee512 is a bit of a beast to get running - I can send you some instructions +files to get CP/M working if you need them.

However, back to non-CP/M mode, if you use `zcc +bee -subtype=wav` a .wav file will be generated which can be passed into ubee512 with ` --tapei=file.wav` I've not figured out how to get the emulator starting up in basic though (I guess I need to select a machine configuration), so falling back to mame this works:

```
mame -window mbeepc85 -cass [wave file]
```

Select basic at the menu.

Type `loadm`, start the tape and it'll load and run.

Fabrizio-Caruso, 2023-07-09:
With nanowasp I get totally wrong position of the characters. Maybe an issue with XSize and YSize:
[image]

This is my Cross-Lib game Bomber compiled with:
```
microbee_no_gfx:
	$(Z88DK_PATH)$(MYZ88DK) +bee $(SCCZ80_OPTS) $(INCLUDE_OPTS) \
	-D__BEE__ -clib=ansi -vn   \
	-D__CONIO_GRAPHICS -D__NO_INIT_GRAPHICS \
	-lndos -create-app -o $(BUILD_PATH)/X$(GAME_NAME)_$@.prg  \
	$(FULL_FILES)
	rm -rf $(BUILD_PATH)/X$(GAME_NAME)_LIGHT_$@.prg
````
or
```
microbee_40col:
	$(Z88DK_PATH)$(MYZ88DK) +bee $(SCCZ80_OPTS) $(INCLUDE_OPTS) \
	-D__BEE__ -clib=ansi40 -vn   \
	-D__CONIO_GRAPHICS -D__NO_INIT_GRAPHICS \
	-lndos -create-app -o $(BUILD_PATH)/X$(GAME_NAME)_$@.prg  \
	$(FULL_FILES)
	rm -rf $(BUILD_PATH)/X$(GAME_NAME)_LIGHT_$@.prg
```
Which options do I need to use to have conio working on this target?

On a general note:
I have used Z88DK for a few years and I love it. Best dev-kit among the ones I use by a large margin.
But I still don't understand what some of the terms used in the doc mean such as "generic console"... and what they imply for the user (which options the user has to use to get what).
I never know which options I have to add because it seems each target has different options to achieve the same result.

suborb, 2023-07-09:
Something somewhere has got the wrong width - it's a bit late to be looking into it today, but I'll do it tomorrow evening.

We've actually got docs about that here: https://github.com/z88dk/z88dk/wiki/Classic-GenericConsole - the key take homes are:

- Both should work with conio
- Generic console is supported on many more targets

Out of interest, which options are you referring to? I think the pragmas have been harmonised, just that for some targets they may not have been implemented (eg setting the startup screen mode). Making things easier is definitely one of the aims - eg console capabilities and features.h were both added with this in mind.

Fabrizio-Caruso, 2023-07-09:
@suborb thanks. As far as the microbee is concerned the problem may be bigger than just XSize, YSize. I will try to reproduce it with a simple test. I think you have a standard conio test that can be used to test whether things like gotoxy are working correctly.

I was refering to things like 
clib=ansi, 
clib=default, 	
-pragma-redirect:fputc_cons=fputc_cons_generic, 
options concerning ix and/or iy, 
etc.
which I only use them for some targets but I often don't understand what I am doing... It is my fault.
I know clib is a way to inform several options, all at once. Sometimes it is need, sometimes it is not needed.
Sometimes I need to tell the compiler to reserve some ix, iy registers but I don't understand why I need to do this when coding in C. Shouldn't this be automatically set by selecting a target?

Fabrizio-Caruso, 2023-07-09:
@suborb for most targets I want to use CONIO, redefine some characters, use bit-banging if available, have either joystick or the most responsive keyboard polling routine available for a target.

Fabrizio-Caruso, 2023-07-10:
I am not so sure I understand the doc you have linked.
Do I need the generic console if I want to redefined characters?
So no ANSI if I want to define characters or UDG?
How do I compile for the generic console?
Is the pragma `-pragma-redirect:fputc_cons=fputc_cons_generic` necessary to set the generic console? If not, what is it needed for?

suborb, 2023-07-10:
[Working on the bee code - I've figured out the weirdness, BASIC runs in a 64x16 text mode which is different to the 80x25 initial mode I've tested against]

> I was refering to things like
clib=ansi,
clib=default,
-pragma-redirect:fputc_cons=fputc_cons_generic,
options concerning ix and/or iy,
etc.

Some of this is historical cruft that's accumulated over the years combined with a desire not to break things for people. And you're one of those people! :)

Originally, we had for example `+ace` which used the ROM the print to screen (with the library ace_clib.lib). Stefano created the VT100 driver and set about creating `+aceansi` as a target that used the VT100 driver to control the screen (with the library aceansi_clib.lib)

Some years later, this was reduced into a single configuration file with two CLIB lines: default, ansi and to use the VT100 terminal you used -clib=ansi

At some point later, everything was folded into a single library (ace_clib.lib) but the command line interface didn't change (still -clib=ansi).

This worked by (internally) using `-pragma-redirect` this allows the pointing of a function in the library to an alternate implementation (in this case we redirected fputc_cons to fputc_cons_ANSI so that every call to fputc_cons() called the ANSI engine)

Around about this time, the VT52 engine (generic console) was implemented, and so we needed to be able to redirect console output to that, so the option `-pragma-redirect:fputc_cons=fputc_cons_generic` became well known.

Then, I got fed up typing these massive lines and implemented aliases, so that on the command line you could type `--generic-console` which mapped into the appropriate options (see https://github.com/z88dk/z88dk/blob/master/lib/config/alias.inc)

At this point it would have been sensible to implement `--vt100` and remove the `-clib=ansi` options but unfortunately we didn't.

There's one machine that is a bit unusual with the -clib option - `+zxn`, because this was added to newlib first of all, so to you classic you have to add `-clib=classic`.

The ix/iy stuff you shouldn't have to care about in most cases - we make the assumption that at least one index register is available for use and the config should swap them around. I imagine there's a few picky machines (zx80,zx81,cpc,vg5k spring to mind) that you might have a work around for.

I think with a couple of additions to features.h we should be able to get a lot of things auto-selecting for you - eg turning on non-firmware keyboard reading

suborb, 2023-07-10:
A lot of the trouble with the bee is in the initial startup mode when coming from BASIC - starting up in 64x16 is a complete mismatch of what was expected.

I've now added code so that -clib=ansi switches to mode 0, -clib=ansi40 switches to mode 2 on startup so we've got the right resolution. So from the next nightly the target should behave more normally.

The Microbee hardware has quite a lot of configurations so although there's basic compatibility between models, something will inevitably break.

The Nanowasp machine doesn't support colour and doesn't have a PCG in RAM. mbeepc85 in Mame does support both of those features. Attempting to set a custom font on the Nanowasp will result in a screen of rubbish.

-----

For what you want to do with your toolkit, I'd recommend that you compile using the option `--generic-console` - this will (hardware permitting) allow you to use a custom font/udgs.

The console driver that you get by default varies - sometimes it's the ROM printer, but otherwise it is the generic console. A quick check shows that the following startup with generic console:

```
ace/classic/ace_crt0.asm:    defc    TAR__fputc_cons_generic = 1
alphatro/classic/alphatro_crt0.asm:    defc    TAR__fputc_cons_generic = 1
coleco/classic/allram.asm:    defc    TAR__fputc_cons_generic = 1
coleco/classic/rom.asm:    defc    TAR__fputc_cons_generic = 1
fp1100/classic/fp1100_crt0.asm:    defc    TAR__fputc_cons_generic = 1
gl/classic/gl_crt0.asm:    defc    TAR__fputc_cons_generic = 1
homelab/classic/homelab_crt0.asm:    defc    TAR__fputc_cons_generic = 1
kc/classic/kc_crt0.asm:    defc    TAR__fputc_cons_generic = 1
krokha/classic/krokha_crt0.asm:    defc    TAR__fputc_cons_generic = 1
laser500/classic/ram.asm:    defc    TAR__fputc_cons_generic = 1
m100/classic/m100_crt0.asm:    ;        defc    TAR__fputc_cons_generic = 1
m5/classic/m5_crt0.asm:    defc    TAR__fputc_cons_generic = 1
mikro80/classic/mikro80_crt0.asm:    defc    TAR__fputc_cons_generic = 1
multi8/classic/multi8_crt0.asm:    defc    TAR__fputc_cons_generic = 1
myvision/classic/myvision_crt0.asm:    defc    TAR__fputc_cons_generic = 1
mz2500/classic/mz2500_crt0.asm:    defc    TAR__fputc_cons_generic = 1
nabu/classic/nabu_crt0.asm:    defc    TAR__fputc_cons_generic = 1
nascom/classic/nascom_crt0.asm:    defc    TAR__fputc_cons_generic = 1
pasopia7/classic/pasopia7_crt0.asm:        defc    TAR__fputc_cons_generic = 1
pc6001/classic/pc6001_crt0.asm:    defc    TAR__fputc_cons_generic = 1
pc88/classic/allram.asm:    defc    TAR__fputc_cons_generic = 1
pencil2/classic/pencil2_crt0.asm:    defc    TAR__fputc_cons_generic = 1
pk8000/classic/pk8000_crt0.asm:    defc    TAR__fputc_cons_generic = 1
pmd85/classic/pmd85_crt0.asm:    defc    TAR__fputc_cons_generic = 1
pps/classic/pps_crt0.asm:    defc       TAR__fputc_cons_generic = 1
pv1000/classic/pv1000_crt0.asm:    defc    TAR__fputc_cons_generic = 1
pv2000/classic/pv2000_crt0.asm:    defc    TAR__fputc_cons_generic = 1
rx78/classic/rx78_crt0.asm:    defc    TAR__fputc_cons_generic = 1
sam/classic/allram.asm:    defc    TAR__fputc_cons_generic = 1
sam/classic/highram.asm:    defc    TAR__fputc_cons_generic = 1
sc3000/classic/sc3000_crt0.asm:    defc    TAR__fputc_cons_generic = 1
smc777/classic/smc777_crt0.asm:    defc    TAR__fputc_cons_generic = 1
sol20/classic/sol20_crt0.asm:    defc    TAR__fputc_cons_generic = 1
spc1000/classic/spc1000_crt0.asm:    defc    TAR__fputc_cons_generic = 1
special/classic/special_crt0.asm:    defc    TAR__fputc_cons_generic = 1
srr/classic/sorcerer_crt0.asm:    defc    TAR__fputc_cons_generic = 1
super80/classic/super80_crt0.asm:    defc    TAR__fputc_cons_generic = 1
sv8000/classic/sv8000_crt0.asm:    defc    TAR__fputc_cons_generic = 1
trs80/classic/trs80_crt0.asm:    defc    TAR__fputc_cons_generic = 1
vector06c/classic/vector06c_crt0.asm:    defc    TAR__fputc_cons_generic = 1
z1013/classic/z1013_crt0.asm:    defc    TAR__fputc_cons_generic = 1
z80tvgame/classic/z80tvgame_crt0.asm:    defc    TAR__fputc_cons_generic = 1
zx/classic/spec_crt0.asm:    defc    TAR__fputc_cons_generic = 1
zxn/classic/ram_crt0.asm:    defc    TAR__fputc_cons_generic = 1
```

However, just be explicit!

Fabrizio-Caruso, 2023-07-10:
@suborb I am not sure I understand. Are UDG broken? The Z88DK says there is support for UDG:
[image]

suborb, 2023-07-10:
UDGs aren't broken, but not all hardware configurations support them. The following models are emulated by Mame. I've no idea what model is emulated by Nanowasp.

```
mbee56            Microbee 64k                              (Applied Technology, 1983)
mbeepc85          Microbee PC85                             (Applied Technology, 1985)
mbee256           Microbee 256TC                            (Microbee Systems, 1987)
mbeeic            Microbee 32k IC                           (Applied Technology, 1983)
mbee              Microbee 16k Standard                     (Applied Technology, 1982)
mbee128p          Microbee 128k Premium                     (Microbee Systems, 1986)
mbeeppc           Microbee Premium PC85                     (Microbee Systems, 1986)
mbee128           Microbee 128k Standard                    (Applied Technology, 1984)
mbeepp            Microbee Premium Plus+                    (Microbee Systems, 2012)
mbeepc85s         Microbee PC85 (Swedish)                   (Applied Technology, 1985)
```

In my testing with `mbeepc85` the following work as expected:

- Colour
- Custom font
- UDGs

My suspicion is that mbee, mbee56 and mbeeic don't support them.

Fabrizio-Caruso, 2023-07-10:
@suborb Thanks a lot for your clarification. Of course thanks a lot for keeping legacy options not to break people's projects.

Fabrizio-Caruso, 2023-07-10:
@suborb what do I need to run mame for the mbeepc85 other than the rom? Do I need a specific boot disk?

suborb, 2023-07-10:
Just the romset is needed.

zx70, 2023-07-11:
The reason of the lack of detail in the documentation is that the ubee512 emulator comes with handy scripts allowing to load any type of binary format (disk image, COM or BEE file) and choosing transparently the proper emulation configurations.

closed 2023-07-12 by Fabrizio-Caruso

Fabrizio-Caruso, 2023-07-12:
@suborb is custom font and UDG the same thing except for the ASCII codes >127 for UDGs?

In the code I do use the code below.
I thought I was using UDGS. The ASCII codes I redefine are below $80. So should I rather say I am using custom fonts? 
```
	void *param = &udgs;
	console_ioctl(IOCTL_GENCON_SET_FONT32, &param);

	int mode = __SCREEN_MODE;

	console_ioctl(IOCTL_GENCON_SET_MODE, &mode); 
```

suborb, 2023-07-12:
> In the code I do use the code below. I thought I was using UDGS. The ASCII codes I redefine are below $80. So should I rather say I am using custom fonts?

That's the distinction that I use - below $80 = font, above $80 = udg

Fabrizio-Caruso, 2023-07-12:
@suborb for debugging it would be much easier to use .BEE files. NanoWasp only emulates an earlier model and redefined characters do not work. The Z88DK documentation says we can load them into UBEE512 with RUNM"A.BEE" but the doc does not mention what we have to do before to attach "A.BEE" to a tape or whatever device RUNM reads from. I have tried to google it with no luck so far. Do you y chance remember how this is done? It seems that UBEE512 is the most accurate emulator.

zx70, 2023-07-14:
There should be some very handy scripts for the UBEE512 emulator.
I think they are part of the support stuff around the emulator.

MB.BAT will automatically configure your system depending on the file extension you are willing to run (.BEE, .COM, etc..)

:: This script allows running a Microbee CP/M, Z80, MWB, BAS program from
:: the command line using uBee512 emulation where the file concerned is also
:: residing on the host system.
::
:: This is very useful for testing programs under uBee512 emulation as it
:: avoids having to manually create a disk image with the required files on
:: it.  Simply specify the Microbee program on the host system command line.
::
:: A special uBee512 boot disk image containing the UBSCRIPT.COM program on
:: the A: image creates '.MWB' and '.SUB' files dynamically to make this
:: possible.  The programs to be run reside on the host system.
::
:: Installing the script and disk image
:: ------------------------------------
:: An up to date 'ubee512_script.ds40' disk image file is required and is
:: available for download from the MSPP Microbee Technology repository:
::
:: http://www.microbee-mspp.org.au/repository
::
:: Windows:
:: The disk image should be located into the same directory that uBee512 is
:: installed.  Place it in the 'disks' directory.
::
:: This script (mb.bat) should be copied to the same directory that uBee512
:: is installed.
::
:: It is required that the environment path points to the uBee512
:: installation location.
::

Fabrizio-Caruso, 2023-07-14:
@zx70 is this valid only for CP/M?

zx70, 2023-07-14:
no, it should choose both the hw and system configuration transparently

Fabrizio-Caruso, 2023-08-03:
@zx70 sort of. I gave up trying to set up UBEE512 to load .BEE files. I use NanoWasp for simple things without graphics.
The Bee target does not have a proper getk, so I am putting it on hold until we can have a getk with auto-repeat. This requires at least the implementation of 2 variants as the way how the keyboard works is different. We should have at least 2 MicroBee models.

This is discussed here:
https://github.com/z88dk/z88dk/issues/923

and on the Microbee forum:
https://microbee-mspp.org.au/forum/viewtopic.php?f=56&t=3017&start=10

zx70, 2023-08-03:
It's a pity the scripts weren't included in the main distribution package, once installed properly the emulator usage is very straightforward