z88dk/z88dk #2354: [bee][bug] conio cursor positioning is broken

Fabrizio-Caruso, 2023-07-10:
Conio cursor position is broken. Tested on Nanowasp.org

This issue is reproduced by:
```
#include <conio.h>
int main()
{
   gotoxy(10,10);
   cprintf("hello world\n");
   gotoxy(10,13);
   cprintf("aui revoir\n");
   return 0;
}
```

compiled with:
```
zcc +bee conio_test.c -create-app
```

suborb, 2023-07-10:
This compilation flags here indicate that this program is using the ROM printer. conio.h only works with `--generic-console`, `-clib=ansi`, `-clib=ansi40` on the Microbee

closed 2023-07-10 by suborb

Fabrizio-Caruso, 2023-07-10:
@suborb Wichh cmpilation flags? You mean that no flags implies rom printer? Maybe you mean that for this specific target, no option implies rom printer.

suborb, 2023-07-10:
Yes. The lack of flags. Sorry for the lack of explicitness.

Fabrizio-Caruso, 2023-07-11:
Now, it looks more broken than before (in NanoWasp) if I just try to display ASCII characters without setting any video.

Characters were shown correctly before and there was just a problem with the cursor positioning.
I might need subtargets or different targets.

My Bomber game is shows a lot of glitched graphics:
[image]