z88dk/z88dk #2355: [bee][bug][ansi40] Wrong position in ansi40 mode

Fabrizio-Caruso, 2023-07-11:
When compiling my Bomber game, the result has wrongly positioned characters in ansi40 mode:
[image]

I am compiling it with
```
microbee_40col:
	$(Z88DK_PATH)$(MYZ88DK) +bee $(SCCZ80_OPTS) $(INCLUDE_OPTS) \
	-D__BEE__ -clib=ansi40  -vn   \
	-D__CONIO_GRAPHICS -D__NO_INIT_GRAPHICS -D_XL_NO_UDG \
	-lndos -create-app -o $(BUILD_PATH)/X$(GAME_NAME)_$@.prg  \
	$(FULL_FILES)
	rm -rf $(BUILD_PATH)/X$(GAME_NAME)_LIGHT_$@.prg
```

suborb, 2023-07-11:
Similar to #2358 - how are you determining XSize?

Fabrizio-Caruso, 2023-07-11:
@suborb I just take whatever default values XSize and YSize have. This works for all other 100 Z88DK targets.
I can try to write a minimal test.

Fabrizio-Caruso, 2023-07-11:
Sorry, you are just right (as usual).
Please excuse me for this stupid non-issue.

closed 2023-07-11 by Fabrizio-Caruso

Fabrizio-Caruso, 2023-07-11:
@suborb So both clib=ansi40 and --generic-console work fine as long as I provide the right values for XSize and YSize.

On the other hand in Nanowasp I don't see redefined characters but I think you mentioned this. It does not crash, though. It just shows the characters as if they were not defined.

Fabrizio-Caruso, 2023-07-11:
@suborb I may open an issue on the input. It seems the keyboard is using an overly long buffer, so that the games become unplayable as long as you keep any key pressed for just a second or less as the player will then just keep on moving on that direction for the next 20 or 30 seconds as if the key were still pressed.