z88dk/z88dk #2361: [bee][feature/imperfection][mode 2] spaces between tiles also in mode 2

Fabrizio-Caruso, 2023-07-12:
The Z88DK platform doc mentions a spacing issue with mode 1 but I am seeing this also in mode 2.
What it be possible to support bigger fonts that would cover those spaces? I guess 8x10 or 8x12.

[image]

Fabrizio-Caruso, 2023-07-12:
@suborb I see that but could you support an option to use 8x10 tiles as it is already the case for targets such as the VG-5000?

suborb, 2023-07-12:
Yes can do, working on the colour for standard at the moment.

suborb, 2023-07-12:
Do you have a set of 8x10 tiles? I had a quick scan through your repo and couldn't spot any.

Fabrizio-Caruso, 2023-07-12:
Not yet but I am working on it. I
In the meanwhile I have implemented the "extend" command in my "xl" script to produce approximate tiles of different shapes from the 8x8 tiles. So that I don't have to redraw them from scratch. This command can be used to do it.

The conversion from 8x8 to approximate 8x10 is already implemented:
[convert ](https://github.com/Fabrizio-Caruso/CROSS-LIB/blob/master/src/xl#L1117)

suborb, 2023-07-14:
I've added support for `-pragma-define:CLIB_FONT_HEIGHT=xx` for modes 0,2 xx=11 and for mode 1, xx=16. Using those values closes up the gap between the rows nicely.

The default is still 8x8 for compatibility reasons.