z88dk/z88dk #2368: Build fails on Ubuntu with garbage path

atirut-w, 2023-07-24:
I did everything exactly as the instructions said, but the makefile seems to contain an error. I'd like to help investigate this, but that would be tomorrow evening. I will leave information here if someone also wants to take a crack at it.

OS: Ubuntu 22.04 WSL

Error sample:
```
mkdir -p -- /usr/local/home/atirut/Projects/z88dk/src/sdcc-build/sdcc-install /usr/local/home/atirut/Projects/z88dk/src/sdcc-build/sdcc-install
mkdir: cannot create directory ‘/usr/local/home’: Permission denied
mkdir: cannot create directory ‘/usr/local/home’: Permission denied
make[3]: *** [Makefile:2800: installdirs] Error 1
make[3]: Leaving directory '/home/atirut/Projects/z88dk/src/sdcc-build/support/sdbinutils'
make[2]: *** [Makefile:2537: install] Error 2
make[2]: Leaving directory '/home/atirut/Projects/z88dk/src/sdcc-build/support/sdbinutils'
make[2]: Entering directory '/home/atirut/Projects/z88dk/src/sdcc-build/doc'
install -d /usr/local/home/atirut/Projects/z88dk/src/sdcc-build/sdcc-install/share/doc/sdcc
install: cannot create directory ‘/usr/local/home’: Permission denied
```

**Info for git bisect**:
Bad commit: bd839f7c659abf9cc2bc868dde20d1e62869a70e
Last known good commit: a09ee3d7a100514c7a1d7c56a1e20955386ab0b7

feilipu, 2023-07-24:
The WSL probably has an issue with the `usr/local/home` path.

Please confirm that you can create that path on your build machine.

If not, then it is clearly WSL related, given the new sdcc v4.3.0 build requirements, and you might need to adjust.

suborb, 2023-07-25:
I‘m AFK at the moment, but we’ve got the same failure with snapcraft. 

I‘ll dig a bit when I’m back later on this week.

atirut-w, 2023-07-25:
> The WSL probably has an issue with creating the `/usr/local/home` path.

I can investigate later today but the build script should *not* be creating anything inside a protected path AFAIK. It seems the build script appended `/usr/local` to a sane path, as evidenced by the username. The compile innstructions does not call for sudo or root access neither.

suborb, 2023-07-25:
Reflecting a bit more - GitHub actions passing, snap and your build failing during an install phase. 

This is probably a tangle up between PREFIX and DESTDIR in the sdcc build. There was a sort out in #1985 but at that point in time we didn’t have to do an install on sdcc - so it got missed. 

As mentioned I’ll investigate when I’ve got a keyboard available.

atirut-w, 2023-07-25:
@feilipu I have managed to repro this on an Ubuntu VM. Definitely an SDCC issue since it doesn't happen if I don't build SDCC.

atirut-w, 2023-07-25:
I have commented on the upstream SDCC issue. Should I close this one? In the meantime, I'll `git bisect` to find the last commit that successfully builds SDCC since I'll need that commit for something of mine.

atirut-w, 2023-07-25:
UPDATE: Nope, this is a Z88DK problem. The root cause was c00b69c2092c885c84ae2f0c207eb499a6c7459b which only changed the Makefile.

feilipu, 2023-07-25:
> [These two lines](https://github.com/z88dk/z88dk/commit/c00b69c2092c885c84ae2f0c207eb499a6c7459b#diff-76ed074a9305c04054cdebb9e9aad2d818052b07091de1f20cad0bbac34ffb52L121-L122) are probably what caused it.

They no longer exist in the `Makefile` as they were improved upon, so they are not the current problem.

But the root cause is related to the new requirement to `install sdcc` and copy the resultant binary for our use.

Check the recent sdcc upgrade PR and Issue to read more background.

atirut-w, 2023-07-26:
Huh, Z88DK builds successfully before that commit, though? Let me double check.

feilipu, 2023-07-26:
> Z88DK builds successfully before that commit.

You're seeing the merge of a PR and the master branch. That commit was in the PR, so when merged it has to go somewhere.

Check the PR #2190 to see what we were doing to try to get sdcc to build properly.

atirut-w, 2023-07-26:
I see. Since this is an issue with SDCC, should this issue be closed? Also, how are you planning to deal with this? Wait it out? Revert SDCC version?

feilipu, 2023-07-26:
Patch the sdcc `Makefile` and advise upstream.

suborb, 2023-07-27:
As I suspected, the install of sdcc is really a fake install so we need to unset `DESTDIR` when doing it.

My bad, I should have checked the actual docker error before merging rather than assuming it was the usual "can't download something" problem.

feilipu, 2023-07-27:
The binary is built in `sdcc/src`.

suborb, 2023-07-27:
It's a dismal sea of red due to #2369, but both snap (x86_64) and docker builds here are now green:

[image]

closed 2023-07-27 by suborb

feilipu, 2023-07-27:
We could even omit the install step entirely, and just copy the binary from `sdcc/src/sdcc`. 

That would also make the prefix and destdir irrelevant.

suborb, 2023-07-27:
Yeah, that was the backup plan - I thought you couldn't find it originally hence the installing?

feilipu, 2023-07-27:
> I thought you couldn't find it originally hence the installing?

I think I thought it was moving things to the right place, and so was a useful step?
Or perhaps it was moving the real binary at one stage, but is no longer...

I can't remember...

anyway now it doesn't do a real copy so the `make install` is not doing anything useful.
I've got a small PR running. Let's see if it passes all the tests.