z88dk/z88dk #2371: Candidate items for v2.3 Release

feilipu, 2023-07-29:
It has been a year, so,...

An open list for things which might be added before the v2.3 Release.
Please add or exclude as relevant or desired...

Or we could open a [project](https://github.com/z88dk/z88dk/projects?type=classic) with the proposed list?

- [ ] Aquarius+ #2352 #2064
- [ ] Parallel build support #2331
- [ ] ez80-clang as compiler #2329
- [ ] homebrew package for v2.3 #1633

For @feilipu 
- [ ] Integer maths Rabbit r2ka 64_32x32
- [ ] math32 Rabbit r2ka optimisation

Would it be best to push #1391 out to v2.4 or a v3.0 (even though it was already in the [v2.2 todo](https://github.com/z88dk/z88dk/projects/4#card-32558393) :wink:), and have a simple Release before tackling the big job?

suborb, 2023-07-29:
Thanks for doing this - it probably is about time again.

* Clang: I think this is done enough for now. There's enough library support to be useful and I've not seen any feedback on it.
* Parallel builds: The immediate problem was sorted, newlib is parallelised, classic should be as well. 
* Aquarius+: I just need to implement `opendir` and friends here.

#1391 should be kicked out again (I'll comment on that issue later)

The one thing that would be nice (but is a big change with possibly lots of issues) is #2320

pauloscustodio, 2023-07-29:
I'm finishing #2338 and will tackle #2320 next.

zx70, 2023-07-29:
There's a couple of issues I totally left behind, which obviously can stay like this a little more.   The TI calculator issue is probably slightly pointless, IIRC the native asm() entry allows very small programs only.
As for the bug on the Aquarius we should find someone owning the hardware and coordinating the tests.

feilipu, 2023-07-31:
> As for the bug on the Aquarius we should find someone owning the hardware and coordinating the tests.

@Fabrizio-Caruso suggested that Roy Templeman (@LoudScotsBloke) might be able to help, if he's listening?

pauloscustodio, 2023-07-31:
> The one thing that would be nice (but is a big change with possibly lots of issues) is #2320

The feature is ready in the issue_2320 branch, but the library does not build, due to incompatible -IXIY options between objects being linked together.

pauloscustodio, 2023-08-01:
On issue #2320: on the issue_2320 branch you cannot include in a library an object file built for a different CPU.
As a possible solution I've added the options "z80asm -m'*' -xlib ,,," that assembles all files for all CPU-IXIY combinations and packs them all in the same library; when linking, z80asm skips the object files for a different architecture and all should work.

zx70, 2023-08-01:
> Hi guys, I'm listening! What would you like me to do?

Roy,
You're very kind showing up immediately!
I left a couple of issues pending related to the Mattel Aquarius, I wonder if it's possible for you to do some testing on the real machine ?

Is redirect you to a proper issue chat

feilipu, 2023-08-05:
@LoudScotsBloke is this issue https://github.com/z88dk/z88dk/issues/2064 still current with the current code?

feilipu, 2023-08-15:
Added math32 r2ka tests... some useful numbers. Even if not fully optimised, I think it will do.

```sh

zcc +test -vn  -DMATH32 -D__MATH_MATH32 -fp-mode=ieee -DMATH_LIBRARY="\"\\\"Math32\\\"\"" -I../../framework -DNO_LOG_RUNNING -DNO_LOG_PASSED  ../../framework/test.c math.c -o test_math32.bin  -lmath32 -m
 21:04:05  Starting test 
Starting suite Math32 Tests (13 tests)
13 run, 13 passed, 0 failed

Ticks: 9959656
 21:04:05  Test completed 
zcc +test -vn -clib=z180  -DMATH32 -D__MATH_MATH32 -fp-mode=ieee -DMATH_LIBRARY="\"\\\"Math32\\\"\"" -I../../framework -DNO_LOG_RUNNING -DNO_LOG_PASSED  ../../framework/test.c math.c -o test_math32_z180.bin   -lmath32_z180 -m
 21:04:06  Starting z180 test 
Starting suite Math32 Tests (13 tests)
13 run, 13 passed, 0 failed

Ticks: 6994100
 21:04:06  z180 test completed 
zcc +test -vn -clib=z80n  -DMATH32 -D__MATH_MATH32 -fp-mode=ieee -DMATH_LIBRARY="\"\\\"Math32\\\"\"" -I../../framework -DNO_LOG_RUNNING -DNO_LOG_PASSED  ../../framework/test.c math.c -o test_math32_z80n.bin   -lmath32_z80n -m
 21:04:06  Starting Z80N test 
Starting suite Math32 Tests (13 tests)
13 run, 13 passed, 0 failed

Ticks: 7618557
 21:04:06  Z80N test completed 
zcc +test -vn -clib=rabbit  -DMATH32 -D__MATH_MATH32 -fp-mode=ieee -DMATH_LIBRARY="\"\\\"Math32\\\"\"" -I../../framework -DNO_LOG_RUNNING -DNO_LOG_PASSED  ../../framework/test.c math.c -o test_math32_r2ka.bin   -lmath32_r2ka -m
 21:04:08  Starting Rabbit test 
Starting suite Math32 Tests (13 tests)
13 run, 13 passed, 0 failed

Ticks: 5335774
 21:04:08  Rabbit test completed 
```

suborb, 2023-12-04:
Things have slowed down again. I’m minded to do a snapshot and release around about the 15th December  - any objections?

suborb, 2023-12-18:
Smidgen delayed on this one (let's say the 15th crept up on me). 

Now the s390 stuff is resolved, I want to get #2461 in and #2462 confirmed as fixed.

suborb, 2023-12-20:
Snapshotted and released.

closed 2023-12-20 by suborb

feilipu, 2023-12-20:
Note to adjust main `README.md` to potato new release.

zx70, 2023-12-21:
Congrats Dom, this autumn you kept parallelizing several issues.   Impressive !