z88dk/z88dk #2372: z88dk built for 32-bit Linux produces erroneous code

dinosaurscanrun, 2023-07-31:
In July 2023 I began to experiment with z88dk to compile C code for a TRS-80 cassette emulation, and for a hardware retrocomputing environment (Z80-MBC2) running CPM/3.  The "classic" sccz80 compiler and library provided adequate cross-compilation facilities and I was pleased with early results, easily producing executables for an "xtrs" emulation of TRS-80 and with minor modifications CP/M *.COM files.  The CP/M targets were found to run well in both the RunCPM emulation environment and also when copied into CP/M disk image files using "cpmtools".  I successfully built the z88dk project using current git sources for both  amd64 and i386 arch variations on Debian 12. I was able to convert the build folder content into *.deb files suitable for installation as a Debian package, complete with a patch I applied to fix an error I found in the asin() function. 

The original asin() function was emitting results exactly half the correct values.  I get the impression from a thread found via Google that this may have been a regression, since I found a prior comment on asin() which suggested the original code had a "2.0 *" term which is NOT in the current sources.

All appeared well in early testing on all platforms until I ventured into testing floating point and various transcendental functions. I found that while the code compiled and ran as expected on the 64-bit (amd64) arch variant of Debian, when I tested the same C source file on a 32-bit host OS, even though all parts of the compilation appeared to work as expected, the actual executable had a glaring error:  a constant entered into the function was changed to another erroneous value (0.25 became -0.125).  Another value entered into the same program was NOT altered (0.7 remained 0.7).

I spent some time examining both the 32-bit and 64-bit downloads for discrepancies, and found no SOURCE discrepancies.  Both directory trees contained identical source files, but they apparently compiled with different functionality under a 32-bit vs. a 64-bit host.

I was able to show that the problem is not limited to my own source build of z88dk by trying the official "snap" packages of z88dk, which were available both for 64-bit and 32-bit hosts.  The identical behavior was found using the snap installations, with the exception of the asin() patch which of course was not applied within the snaps.  Whatever the issue is, it is endemic to the z88dk project and not to my own building technique.

I investigated a little further by examining the ASM interim products used to create the ultimate target executable.  When compiled to ASM target instead of app target, the ASM clearly shows a comment (taken from C source) for the storing of a constant 0.25, but in the 32-bit product the 6-byte value is WRONG.

From 64-bit host ASM product:

.i_2
	;0.250000
	defb	0x00,0x00,0x00,0x00,0x00,0x7f  ; This is actually a correct value for 0.25
.i_3
	;0.700000
	defb	0x33,0x33,0x33,0x33,0x33,0x80  ; This is actually a correct value for 0.7

From 32-bit host ASM product (same source code):

.i_2
	;0.250000
	defb	0x00,0x00,0x00,0x00,0x80,0x7e  ; NOTE the 5th byte shows 0x80, including the negative sign bit
	                                       ; The value is actually interpreted as -0.125, NOT 0.25
.i_3
	;0.700000
	defb	0x33,0x33,0x33,0x33,0x33,0x80  ; This is actually a correct value for 0.7

Examination of the temporary (created and destroyed under /tmp) ASM products indicate the issue apparently arises during the first optimizer step, but I lack resources to explore further.  

I like your project and hope it continues to succeed.  However, I would also like it to work correctly on a 32-bit host.

Below I summarize the test inputs and results.

Patch applied to original asin() function in build of 7/30/2023:
-----------------------------------------------------------------
z88dk/libsrc/math/cimpl/asin.c:

/*
 *	asin(x)
 *
 *	-1 < x < 1
 *
 *	Undefined results otherwise
 *
 *	$Id: asin.c,v 1.1 2008-07-27 21:44:57 aralbrec Exp $
 */

#include <float.h>
#include <math.h>

/* Patched to fix library error 20230730 GLF -- apply 2.0* in front of original */
double asin(double x)
{
	return 2.0*atan(x/(1.+sqrt(1.-(x*x))));
}

Source code to test float issue and illustrate patch for asin():
----------------------------------------------------------------

#include <stdio.h>
#include <math.h>

float x;

main()
{
  x = 0.25;

  printf("\fTest of various floating point functions\n");
  
  printf("x = %f  (should be .25)\n\n",x);
  printf("sin(x) = %f  (correct: .247404)\n",sin(x));
  printf("tan(x) = %f  (correct: .255342)\n",tan(x));
  printf("sqrt(x) = %f  (correct: .5)\n",sqrt(x));
  printf("asin(x) = %f  (correct: .252680)\n\n",asin(x));

  x = 0.7;
  
  printf("x = %f  (should be .7)\n\n",x);
  printf("sin(x) = %f  (correct: .644217)\n",sin(x));
  printf("tan(x) = %f  (correct:  .842288)\n",tan(x));
  printf("sqrt(x) = %f  (correct: .836660)\n",sqrt(x));
  printf("asin(x) = %f  (correct: .775397)\n\n",asin(x));
}

Compiler command line (either 32-bit or 64-bit Debian 12 host):
---------------------------------------------------------------
zcc  +cpm -O2 -create-app -lm testf.c -o testf.com

Example run of TESTF.COM compiled using sccz80 classic library on Debian 12 amd64 OS:
-------------------------------------------------------------------------------------

(NOTE: asin() function was patched 7/30/2023 to produce correct result)

  CP/M Emulator v6.1 by Marcelo Dantas
      Built Jul 18 2023 - 00:52:49
----------------------------------------
BIOS at 0xfe00 - BDOS at 0xfd00
CCP INTERNAL v3.0 at 0xfd00

RunCPM Version 6.1 (CP/M 60K)

A0>b:
B0>dir
B: CMD2     COM  |  TEST     COM  |  TESTF    COM
B0>testf
Test of various floating point functions
x = 0.250000  (should be .25)

sin(x) = 0.247404  (correct: .247404)
tan(x) = 0.255342  (correct: .255342)
sqrt(x) = 0.500000  (correct: .5)
asin(x) = 0.252680  (correct: .252680)

x = 0.700000  (should be .7)

sin(x) = 0.644218  (correct: .644217)
tan(x) = 0.842288  (correct:  .842288)
sqrt(x) = 0.836660  (correct: .836660)
asin(x) = 0.775397  (correct: .775397)

RunCPM Version 6.1 (CP/M 60K)

B0>

Example run of TESTF.COM compiled using sccz80 classic library on Debian 12 i386 OS:
-------------------------------------------------------------------------------------

(NOTE: asin() function was patched 7/30/2023 to produce correct result)

  CP/M Emulator v6.1 by Marcelo Dantas
      Built Jul 26 2023 - 20:13:19
----------------------------------------
BIOS at 0xfe00 - BDOS at 0xfd00
CCP INTERNAL v3.0 at 0xfd00

RunCPM Version 6.1 (CP/M 60K)

A0>b:
B0>dir
B: CMD2     COM  |  TEST     COM  |  TESTF    COM
B0>testf
Test of various floating point functions
x = -0.125000  (should be .25)

sin(x) = -0.124675  (correct: .247404)
tan(x) = -0.125655  (correct: .255342)
sqrt(x) = 0.000000  (correct: .5)
asin(x) = -0.125328  (correct: .252680)

x = 0.700000  (should be .7)

sin(x) = 0.644218  (correct: .644217)
tan(x) = 0.842288  (correct:  .842288)
sqrt(x) = 0.836660  (correct: .836660)
asin(x) = 0.775397  (correct: .775397)

RunCPM Version 6.1 (CP/M 60K)

B0>

suborb, 2023-08-01:
So to summarise, there's two issues:

- `asin()` generates incorrect results
- On 32 bit Intel, some floating point values aren't resolved correctly (possibly negative powers of 2?)

I've resolved the issue with `asin()` - a rogue `1.+` had entered the implementation. 

With the other issue, I can't reproduce it at the moment. I've tried on arm32bit as well as using -m32 on a 64 bit Intel box.

suborb, 2023-08-01:
It's a bit of a Heisenbug - it occurs when sccz80 is compiled with -O2.

Now, to figure out what's changed...

dinosaurscanrun, 2023-08-01:
Thanks for addressing the asin() issue.  It made an otherwise excellent
package look bad to produce incorrect numbers, especially if they might
come up a lot, especially in game programming.

As for the 32-bit thing, I'm sorry to present a difficult-to-reproduce
issue.  Most developers these days seem to have moved on from 32 bit OSes
and exclusively use 64-bit environments.  If they want to deal with 32-bit
code they acquire some kind of mulitiarch capability (Linux) or have other
32-bit support in the OS (WoW32).  That's not my situation.   I tend to
support a raft of what would be considered obsolete machines (partly hobby,
partly nostalgia, but so is anything Z80). Some of my machines do not have
a 64-bit CPU, but I can still run current Debian OS on them (slowly).

I actually "roll my own" Debian Linux live boot remixes, so I have
available current Debian 12 i386 arch live bootable builds to test on.
Debian still supports 32-bit repositories and kernels, and almost all
Debian applications are still available as either 32-bit or 64-bit.
 Debian officially seems to have suspended production of 32-bit live
bootable versions for x86, but they do provide installer sets for 32-bit.
I personally prefer the live boot versions, especially for testing
software. but trust me, they can still be built using official Debian
resources (live-build).

When I referred to a 32-bit host, I meant that the z88dk package was
compiled on and intended to run on a 32-bit i386 (686) Linux OS.  I have
not experimented with -m32 under a 64-bit host.  In my experience, most
applications compiled form source with GCC seem to produce equivalently
functioning results whether the host is 64 bit or 32 bit, unless some
feature is looking specifically for a 64-bit  feature unavailable in the
32-bit OS.

This might be a niche situation for most developers.  However, it bothers
me that the z88dk Makefile uses the identical source code for both 64-bit
and 32-bit builds, and the entire Z80-side source trees (at least
pre-compilation) are identical, but the compiler components behave
differently on the two types of OS. Also, for what it's worth, i did try
every optimization level -O0 thru -O3, and the problem persisted.

I understand there may not be a hurry to address this, given the low number
of 32-bit OS users.  But since the problem does show up in the "snap"
packages for 32-bi Linux, I thought somebody in the thick of things should
know the code was coming out wrong for floats.

Anyway, thanks again for the prompt response on asin() -- I think this is
the first time I actually tried to submit a suggested patch for a current
open source project, and the response was gratifying

suborb, 2023-08-02:
I've merged a fix for this issue (or at least what I managed to reproduce). Please pull/rebuild and a try again and report back either way!

I need to figure out a way to remove the i386 snap - with Ubuntu no longer releasing the build can't be run and is very old.

dinosaurscanrun, 2023-08-02:
WOW!!!!

I must say I'm impressed with your team.  I just finished rebuilding z88dk
(it takes me 20 minutes) as an i386 build on an i386 host OS, and
EVERYTHING WORKED!

I have yet to test build it for amd64, but you successfully addressed
both issues i brought up for i386 build.  I am particularly impressed by
not only te speed with which you fixzxed the issue, but with the way you
knew your product well enough to incisively cut right to the nut of both
problems.  Not only that, but I have to believe you provided support to
what has to be a tiny percentage of your user base -- the users of 32-bit
Linux on PC.

From the test (CP/M target) run on a live booted 686-based Linux OS:

B0>testf
Test of various floating point functions
x = 0.250000  (should be .25)

sin(x) = 0.247404  (correct: .247404)
tan(x) = 0.255342  (correct: .255342)
sqrt(x) = 0.500000  (correct: .5)
asin(x) = 0.252680  (correct: .252680)

x = 0.700000  (should be .7)

sin(x) = 0.644218  (correct: .644217)
tan(x) = 0.842288  (correct:  .842288)
sqrt(x) = 0.836660  (correct: .836660)
asin(x) = 0.775397  (correct: .775397)

BOTH problems fixed -- the 0.25 is correct (was -0.125 in the bad version)
and the offoicial asin() now is correct.

Kudos!

The only sad part of this for me, is that you are considering dropping the
32-bit snap package rather than fixing it.  I am not personally a fan of
snaps (I do use AppImages) but the availability of the z88dk snaps was a
valuable part of my diagnosis of this issue, since it provided a way to
isolate any build issues I might have had.  I have been fighting what
sometimes feels like a one-man crusade to keep the 686 platform viable for
Linux, and i hate to see yet another deprecatio of 32-bit support.  I had
to successively give up a few years ago the ability to boot a modern Linux
on a 486, then on a 586, as the distributions either dropped spport
intentionally or included oackages which required a minimum of 686 feature
set to run.  I ran some tests last week -- the oldest machine I have that
can boot and run Debian 12 is from 2002.  Somewhere I have a Cyrix 586
class machine running an OLD version of Knoppix wich I can no longer update
or replace.  Sigh.

dinosaurscanrun, 2023-08-02:
Followup to prior post...

In the previous oost, I verified your patches worked for the 32-bit build of z88dk. I followed up that report and was able to rebuild and test under my remix of Debian 12 MATE (amd64) and can confirm the patches also work for 64 bit. 

Also, in the interest of suggesting a possible reproducible build environment to further test for PC 32-bit, I retried the z88dk i386 build using the LAST KNOWN official Debian Live for i386 on the web, from the following URL:
   https://cdimage.debian.org/mirror/cdimage/archive/11.7.0-live/i386/iso-hybrid/

For my second build attempt for i386, I downloaded the Debian MATE 11.7 i386 hybrid ISO.   Debian 11.7 is the last update for the "old stable" prior version (bullseye) which has been superseded this year by Debian 12 (bookworm). The Debian 11 repositories and other support are still available online. The current Debian 12 Live site only hosts amd64 ISO images. Though they recommend upgrade to 12, Debian 11 is still working for now, for both amd64 and i386.  I was able to boot from USB using the 11.7 i386 ISO, and after installing git and a couple of other utilities, I was able to install the dependencies and run your build script from an external USB EXT4 work drive (live boot shares RAM with "hard drive" RAMDISK, and the work drive prevents running out of RAM).

The results using today's z88dk patches worked fine, again fixing both reported issues. Thanks again for your excellent work.

I do post my remix ISOs to a (cheap) Google Drive site, if you're interested. I have reason to believe my custom remixes of Debian 12 MATE are the ONLY such live bootable images for i386 (686) platforms currently available on the web.  I don't know if some other non-Debian OS still posts CURRENT i386 arch live ISOs, but they seem to be dying off quickly. Debian 12 does post INSTALLER disks images for i386, but not live bootable images. 

If you're interested,  my "CustomOSMaker" custom remixes of current Debian 12 MATE live images (amd64 and i386):
   https://drive.google.com/drive/folders/1MZD8p23QGoJz_2GVZTkcUBT-hpD_Pm4A?usp=drive_link
   
NOTE: to download from Google Drive may need to temporarily enable 3rd party cookies

suborb, 2023-08-02:
Thanks so much for confirming that it's resolved for you.

> The only sad part of this for me, is that you are considering dropping the 32-bit snap package rather than fixing it.

We've not got much choice, we use snapcraft.io as a (free) way of building and testing on different architectures. It's only core/18.04 that supports i386 and that went out of support at the end of June.

closed 2023-08-02 by suborb

feilipu, 2023-08-02:
> Also, in the interest of suggesting a possible reproducible build environment to further test for PC 32-bit, I retried the z88dk i386 build using the LAST KNOWN official Debian Live for i386.

It would be useful to have at least the annual release available as a i386 binary. I don't know if you would be happy to build (and deb package) for debian i386 at least?

I'm sure we could host and link to them somewhere public, if you like.

dinosaurscanrun, 2023-08-03:
I'm open to the idea, but my methods for building a *,deb file are hacked
together at best.  I have some skills but I've never had to officially
maintain a Debian package for others.  I mostly figure out puzzles.  I use
Debian live-build to put together working live boot Linux OS images from
(mostly) repository prebuilt binary packages.  I have eclectic tastes in
the packages I want to use, and I write programs largely to solve problems
of personal interest. The advantage of making my own OS remix is that I can
make a working environment with all the facilities, libraries,
tools,etc.needed to build and run. If I write a program which somebody else
can use, I want to be able to turn over not just the source code, but a
working build and test environment so they can actually maintain it. I also
find I can completely build from scratch an up to date Linux OS in far less
time than it takes to run a month's worth of Windows Update.

In doing so, I find it practical to include *.deb files for packages which
are useful to me but aren't in the official repositories, like z88dk.  I do
have other "unofficial" options. Since I have control over the build
process, I can do things at build time in the scripts that say, compile
from source and install to strange places without using "correct" means.  I
also liberally override variables and config files with hand-cobbled
replacements.  Using "./configure, make, sudo make install" usually works
out, but sometimes the .configure doesn't work and I only sometimes get
that to work.  Cmake is great when it works, and confusing as hell when it
doesn't. If I can get an app or library to build from source at all, I can
sometimes get it into a *.deb file and include it in my own OS build.

Over the years I've learned just enough about how to make *.deb files work
to be dangerous.  I have never quite figured out how the official
packages are put together with all their myriad connections to other
libraries and source code. How to debbuild?  I only ever found one package
with instructions on how to do that and it just worked without me
understanding how..  I have gotten "checkinstall" to work SOMETIMES, and I
have been able to unpack a *.deb file to a folder, edit it manually, and
pack it back into another *.deb file with "dpkg-deb -b".  It really depends
on whether the actual developers provide a source package I can figure out
how to build. I have not yet figured out how to get "postinst" scripts to
do everything they're supposed to do, so I hack a workaround in the OS
build as needed.

At least the instructions for building z88dk from "git clone" were
available, and appear to work. I use "export BUILD_SDCC_HTTP=1", because
the subversion alternative hasn't worked for me,

However, what I have done with z88dk so far only scratches the surface.
Given that the targets I have built with iit are for effectively obsolete
and personally available platforms, I have only tried sccz80 and classic
library.  I have yet to successfully use sdcc for anything, so I wouldn't
know if I messed it up.  I found an interesting side effect of the z88dk
build that if I tried to avoid building sdcc at all (just omit the flags),
something somewhere broke in sccz80 and its compiled products didn't work
the same (at least for console input).  My "workaround" was to build the
whole package anyway, in which case the problem went away.  I still don't
know why.  But that does mean I have built the whole thing, I just haven't
tested the whole thing.

Because I have avoided sdcc, the *.deb packages I have built so far for
z88dk only copy into the package the necessary files I have found to
make sccz80 work (to the extent i know about).
The sdcc support is extensive, but it takes up a HUGE footprint.  By
omitting the "libsrc" folder alone I seem to be saving about a gigabyte of
installed content.  I appreciate that you have put a lot of work getting
that stuff to work and it represents the bulk of your attention.  I am at
this time unqualified to evaluate it or even to know if I installed it
correctly.

I will make an effort in the next few weeks to see if I can figure out
enough about sdcc and the rest of your package to make a viable installer
for the whole thing.  If so, we can revisit this. I am already making the
build effort anyway, and would be glad to share.  I must warn you though,
that I cannot provide a long-term solution for this problem.  I am willing
to help if I can, but I am 70 years old, and have some health issues, so
anything I start would need somebody else to learn to follow up.

Anyway, I am gratified that you think I could contribute.  You have an
impressive pile of useful stuff in z88dk, and your team seems to be on top
of a lot of things I barely know about.  Every time I see somebody
supporting the "old stuff" like Z80, it validates the attention I gave to
personal computing in my youth, and it feels like some of the best parts
will survive.  My GitHub moniker, "dinosaurscanrun", embodies this -- the
old stuff ain't dead yet.

feilipu, 2023-08-04:
> I will make an effort in the next few weeks to see if I can figure out enough about sdcc and the rest of your package to make a viable installer for the whole thing. If so, we can revisit this. I am already making the build effort anyway, and would be glad to share.

Actually, the `sdcc` build can be done completely separately, just following the instructions for patching, configuring, and building found in the Makefile. The resulting `sdcc` binary is then just lifted into the right `z88dk/bin` directory as `z88dk-zsdcc`.

> I must warn you though, that I cannot provide a long-term solution for this problem. I am willing to help if I can, but I am 70 years old, and have some health issues, so anything I start would need somebody else to learn to follow up.

No stress. As a once off per annum, I'm sure that one of us can build on 32- bits using debian 11 or debian 12. So have a look if you like, and please advise on any gotcha's that you see. We can pick it up later as part of annual releases.

Cheers, Phillip

dinosaurscanrun, 2023-08-04:
OK.  I actually did try to build the full package last night on amd64.
After a couple of failed tries, I think I got a workable *.deb installer
for everything, but it's a whopper.  The installers I had been using which
omitted sdcc were about 9 Mb *.deb files.  The full version is currently 71
Mb.  I can't even get gdebi (the GUI deb installer) to install the big
one.  I was able to get it in with "sudo apt install", but it consumes
almost 2 Gb on the machine, which is about the limit I can handle on a live
boot.

I probably do not have the minimal or even optimal package being
installed.  Because I don't know everything the package wants at runtime, I
threw the kitchen sink at it -- I used all the build dependencies from your
script, not knowing which are build time dependencies and which runtime.  I
tested the install running a version of Debian 12 which had a minimal
footprint -- my usual working OS has a lot of programming libraries and
tools built in and I needed to know that the base installer would work for
such a minimal OS.  That was how I found out that some of the dependencies
I had previously chosen to omit were actually needed, so I put the whole
list back in.

Because I have not previously tried sdcc, I need to learn some more about
newlib to fully adapt and test it.  It seems to have some functions missing
I usually expect, like stdio.h stuff.  I was able to get the same "TESTF.COM"
to run on the CP/M emulator, and it seemed to work.  The binary was smaller
than the sccz80 one, too.  I did need to put in the "#pragma printf %f" for
that one.

I can appreciate that the sdcc compiler itself may have improved features
and be more standards-conformant, but since my needs were for "toy"
applications on obsolete and emulated platforms, I was happy to get ANY C
working.  But if the bulk of your user base is focused on sdcc, I need to
fit any installer to those needs.

Anyway, I need to try out some more complete tests before I'm comfortable
that I didn't leave something out.  I also want to try installing it on
Ubuntu and maybe Mint to see if it still works.  I don't usually use those
but I know where to get the live images, at least for 64-bit.

Because I saw that I needed at least SOME files from z88dk/llbsrc and
z88dk/src, I included everything from those in the installer.  If you can
assure me of parts that aren't needed at runtime, I can omit them
surgically if it saves significant space.  Is Perl and/or all those Perl
components used at runtime (for the compiler and tools) or are they only
used for building the package?

I also need to verify that my much smaller installer for
sccz80-only doesn't leave out something I haven't run into yet.

I apologize. To some extent I'm thinking out loud in this thread.  Adapting
to your package has actually been simpler than many I've tried to
build, and it does actually WORK when I've built it. It's just that it has
so MUCH in it, it's sort of like stumbling across an Einstein-Rosen bridge
into a parallel universe, and I'm still exploring the new world.  It may be
a while before I'm comfortable that I'm not messing up.

I'll keep posting progress on this thread as it happens.

suborb, 2023-08-04:
> Because I have not previously tried sdcc, I need to learn some more about newlib to fully adapt and test it. It seems to have some functions missing I usually expect, like stdio.h stuff. I was able to get the same "TESTF.COM" to run on the CP/M emulator, and it seemed to work. 

It's important to note that classic fully supports sdcc - just add -compiler=sdcc to the command line you used for sccz80.

For more fun (after a lot of building), you can use -compiler=ez80clang to use clang (again with classic) [https://github.com/jacobly0/llvm-project]

dinosaurscanrun, 2023-08-06:
I'll hold off on clang for a while -- I'm still trying to figure out sdcc.
I'm not ready to jump from GCC anyway for any of my other work.

News to report -- I have been pursuing two lines of investigation.  One,
can I get a *.deb file to install on systems foreign to my normal working
environment.  Two, can I better minimize its footprint when sdcc is built
in.

I downloaded several live boot alternative OS ISOs, admittedly all founded
on Debian -- Ubuntu desktop, Ubuntu Mate, Linux Mate.  I did find a Centos
7 (old) ISO which was built for i386.  I'd like to test that ( I think I
can convert the deb file to rpm), but I have no experience with Centos, so
that will wait..

Because I am a daily user of Debian Mate (my remixes are based on that) I
was most familiar with those.  Ubuntu Mate and Linux Mint Mate installed
z88dk OK -- the deb file loaded as expected (sudo apt install).  I couldn't
make the (pushed) "Ubuntu Desktop" install z88dk -- it acted like it
couldn't find most of the dependencies in live mode (I personally hated
that UI anyway). I have yet to test the newer "Mantic" Ubuntu Cinnamon
image, which is newer than the others based on Ubuntu "Jammy" 22.04 LTS.
All of these tested the 64-bit amd64 arch.  I already know I can get the
i386 version to work on Debian.

On minimizing the footprint -- I examined what files and folders seemed
to only be needed at build time vs. those needed at runtime.  I have a
problem with build errors (perl modules?) when attempting to run the "sudo
make install" after building, so I have been forced to manually set up an
installation package.  Originally I took the "kitchen sink" approach and
included everything, not knowing what could be omitted.  However the sheer
size of the sdcc install forced me to look closer at what was truly
needed.  I couldn't figure out what was installed from just looking at the
makefile, so I just experimented by trial and error.  If I took out
something and it broke, I put it back in.  I did explore the file names and
their contents to look for clues -- what referred to *.pl files, what used
*.m4 files, etc.

I came to the conclusion that I could avoid perl and its modules as
dependencies because they seemed to only apply during compiler build time.
I left the perl files in because they didn't consume much room.  I also was
able to eliminate most of the content of the src folder, except for a
couple of config and m4 files.  That saved a LOT.  The amd64 arch *.deb
file went from 70 Mb to 33.5 Mb, and the RAMDISK taken up went from 2.5 Gb
down to 1.8 Gb.  So far it looks like everything I've tried with either
sccz80 or sdcc has worked with the current installation set, and its source
is up to date as of today's download.

I may try to split the installation to allow an sccz80-only install
separate from the full install because it takes up so much less space, but
I really need to spend some time coming up with a lot more test code
(convert some more of my existing C code from other platforms).  Until I am
more familiar with what each compiler and library has to offer, I won't be
comfortable that my installer is ready to be shared outside my own OS
environment.  Give me a month or so.