z88dk/z88dk #2377: (toolchain) Support r800

suborb, 2023-08-08:
We might as well complete the set:

- SLL doesn't exist
- Index halves supported
- New multiplication instructions (MULUB does unsigned 8bit multiplication, and the MULUW instructions does 16bit unsigned multiplication):

```
* ED F9 | MULUB A,A
ED C1 | MULUB A,B
ED C9 | MULUB A,C
ED D1 | MULUB A,D
ED D9 | MULUB A,E
* ED E1 | MULUB A,H
* ED E9 | MULUB A,L
ED C3 | MULUW HL,BC
* ED D3 | MULUW HL,DE
* ED E3 | MULUW HL,HL
ED F3 | MULUW HL,SP
```

(*) indicates that they may not work properly.

When using the MULUB instructions, the result is always stored to HL. While using the MULUW instructions the result is always stored to DE:HL

feilipu, 2023-08-08:
> New multiplication instructions (MULUB does unsigned 8bit multiplication, and the MULUW instructions does 16bit unsigned multiplication).

Interesting. I've been procrastinating wrt Rabbit because it does signed multiply which I believe is annoying for chained (wide) multiply operations. But the unsigned ops sound more useful.

But what does "* may not work properly" mean?

Do you know whether it is possible to get a R800 anywhere besides in a MSX?

suborb, 2023-08-08:
The Japanese reference here: https://ia601300.us.archive.org/34/items/MsxTurboR/MsxTurboRTechnicalHandBook.pdf

More info in this thread: https://www.msx.org/forum/msx-talk/development/uncovering-the-r800

suborb, 2023-08-08:
> But what does "* may not work properly" mean?

No (from R800 user guide): 

[image]

> Do you know whether it is possible to get a R800 anywhere besides in a MSX?

I think that was the only place it was used.

pauloscustodio, 2023-08-12:
Closed by Github; only z80asm and ticks done.

suborb, 2026-11-08:
I'm going to close this one - we have an r800 library and 80cc supports it. MSX now has a `--r800` alias which enables the optimisations.

commit 1ce38ea0 (2023-08-09): Fix #2377 (z80asm) add support of r800
closed 2026-11-08 by suborb