z88dk/z88dk #2381: error: file not found: r2ka_crt0.lib

pauloscustodio, 2023-08-15:
The error:
```
error: file not found: r2ka_crt0.lib
```
started popping up when calling:
```
zcc +test -clib=rabbit -otest_t_issue_0343.bin test_t_issue_0343.c
```
The C file can be a simple ```int main(){}```

feilipu, 2023-08-15:
Yes. The root cause is library name changed from `r2k_crt0.lib` to `r2ka_crt0.lib`.

I guessed it was something to do with Windows using the yesterday daily to do test where the file doesn't exist yet. But I am not sure.

I also couldn't find the `.c` file generating the error. It is created by Perl I guess?

pauloscustodio, 2023-08-15:
Yes, windows cannot generate the libraries and uses the latest nigthly. The error message is fom z80asm while trying to link. The test .c file is creted by the Perl test script and then removed.