z88dk/z88dk #2382: (z80asm) not detecting dupplicate definition on command line

pauloscustodio, 2023-08-18:
test case: z88dk-z80asm -Dabc -Dabc test.asm

pauloscustodio, 2023-08-18:
Error message in future z80asm (branch z80asm2)
```
TARGET=krokha TYPE=8080 z88dk-z80asm -d -I/home/runner/work/z88dk/z88dk/lib/config//../ -DSTANDARDESCAPECHARS -m8080 -DFORkrokha -x/home/runner/work/z88dk/z88dk/libsrc//krokha_clib @/home/runner/work/z88dk/z88dk/libsrc//../libsrc//target/krokha/krokha.lst

--- Building Lambda 8300 Maths Library ---

make --no-print-directory -C math/zxmath mlambda
z88dk-z80asm -I../ -I/home/runner/work/z88dk/z88dk/libsrc//../lib/ -IXIY -DFORlambda -D__CLASSIC -IXIY -DFORlambda -Oobj/mlambda c/sccz80/acos.asm
error: duplicate definition: FORlambda
```

suborb, 2023-08-18:
Should it only error if the definitions are different?

pauloscustodio, 2023-08-18:
> Should it only error if the definitions are different?

That's the behaviour of gcc; and if different it issues a warning.

commit ab856668 (2023-08-18): Fix #2382 (z80asm) report duplicate definition on -D

if different values are given
commit 606b7a21 (2023-08-18): Fix #2382 (z80asm) report duplicate definition on -D

if different values are given
closed 2023-08-18 by pauloscustodio