z88dk/z88dk #2383: Compile built-in functions

dikdom, 2023-08-22:
Hello, 
I have a short (but very informative) discussion with @feilipu  under his z88dk-libraries repo. 
My last two question remained unanswered, could someone pls help me out here/there? (https://github.com/feilipu/z88dk-libraries/issues/2#issuecomment-1687296938)
- is the IX register usage _silently_ replaced with IY during compile/link time when the -mz80_ixiy switch used or an intermediate asm file is generated here also? 
- is there a way to make the zcc (or its used compiler/linker/etc..) to keep this intermediate .asm file (even in tmp dir) and leave there after a successful build?

suborb, 2023-08-24:
Sorry, I'm having a slightly busy week.

It's a silent switch made at assembly time (with a `0xdd <-> 0xfd` made in the object file). Note issue #2320 prevented mixing object files with different -IXIY options .

The intermediate files can be left if you supply `-no-cleanup` *but* there's a lot of files. It's probably easier to split the generation into multiple steps if you care about the .asm file.

```
%.o: %.asm
    zcc ... -o $@

%.asm: %.c
   zcc ... -o $@
```

feilipu, 2023-09-06:
Closed with https://github.com/feilipu/z88dk-libraries/issues/2

closed 2023-09-06 by feilipu