z88dk/z88dk #2386: scanf %[] sequences not working

dfandrich, 2023-09-04:
There seems to be code to handle this, but it's not working as shown in this test program:
```
#include <stdio.h>

int main(void) {
    int rc;
    int i = -1;
    char buf[10] = "";

    rc = sscanf("xxxzzzz 123", "%*[^ ] %d", &i);
    printf("rc: %d i: %d\n", rc, i);

    rc = sscanf("xxxzzzz 321", "%*[xz] %d", &i);
    printf("rc: %d i: %d\n", rc, i);

    rc = sscanf("xxxzzzz 321", "%[xz] %d", buf, &i);
    printf("rc: %d buf:%.10s i: %d\n", rc, buf, i);

    return 0;
}
```
This returns the following under gcc on Linux x86_64 (i.e. the character sequences are properly scanned):
```
rc: 1 i: 123
rc: 1 i: 321
rc: 2 buf:xxxzzzz i: 321
```
But under a CP/M build with z88dk it returns failures for each scanf:
```
rc: -1 i: -1
rc: 0 i: -1
rc: 0 buf: i: -1
```

suborb, 2023-09-04:
What compile line are you using?

I don’t remember writing code to handle this,  but it was a few years ago.

dfandrich, 2023-09-04:
The test program was compiled with: zcc +cpm -create-app scanfail.c
I don't recall exactly how I compiled z88dk, but I just followed the
instructions (as of a couple of weeks ago).
z88dk/libsrc/_DEVELOPMENT/stdio/z80/input_helpers/__stdio_scanf_bkt.asm looks
like it should be handling this, but I'm not positive this is the code in the
libc zcc is giving me.

suborb, 2023-09-04:
That's the newlib version, the version you're using with your compile line is in this directory: https://github.com/z88dk/z88dk/tree/master/libsrc/stdio

So the code will need porting back.

Sorry!

suborb, 2023-09-07:
Hopefully that'll work for you - I've still to write some UTs for it, but your example now works.

dfandrich, 2023-09-08:
That fixes my application. Thanks!

suborb, 2023-09-09:
Thanks for the confirmation.

closed 2023-09-09 by suborb