z88dk/z88dk #2388: Rabbit 4000/5000 support

suborb, 2023-09-07:
How do you feel about this @pauloscustodio? 

From the z88dk perspective, the main thing needed is support for the 0x7f page where some of the main page opcodes have been moved.

pauloscustodio, 2023-09-08:
Let me have a look during the weekend.

pauloscustodio, 2023-09-08:
Can you please provide a link to the assembly specs? A quick Google search only retrieved documents on the HW or programming with Dynamic C.

feilipu, 2023-09-08:
https://github.com/z88dk/techdocs/blob/master/rabbit/Rabbit%20Instruction%20Reference.pdf

pauloscustodio, 2023-09-08:
Thanks, @feilipu

suborb, 2023-09-16:
z88dk-dis should now support Rabbit 4000/5000, though there's bound to be a few bugs!

pauloscustodio, 2023-09-16:
Thanks, @suborb. I'll plug it in the z80asm test process to detect bugs in both ticks and z80asm. I'm still half-way through the 4000/5000 opcodes.

pauloscustodio, 2023-09-17:
Does the Rabbit have the PO and PE flags? In page 93 of the [Instruction Reference](https://github.com/z88dk/techdocs/blob/master/rabbit/Rabbit%20Instruction%20Reference.pdf) they are replaced by LZ and LO flags, but in the summary tables on page 400 and 401 they appear instead of LZ and LO.

suborb, 2023-09-17:
I think PV is a fat finger for LV

pauloscustodio, 2023-09-17:
Ok, I'll remove PO and PE from all the Rabbits.

pauloscustodio, 2023-09-17:
> Ok, I'll remove PO and PE from all the Rabbits.

Bad idea! The library is full of jp po/pe meaning nv/v. Reverting to accepting po/pe, but keeping change in disassembler to output lz/lo

suborb, 2023-09-17:
To avoid duplicate work I’ve got a branch with ticks support for the 4k - I’m focussing on the main page + the moved 7f page.

suborb, 2023-09-18:
That's now merged in. In terms of the main page the following aren't implemented:

- [ ] 0x54: xor hl,de
- [ ] 0x45 sub hl,jk
- [ ] 0x55 sub hl,de
- [ ] 0x65 add hk,jk
- [ ] 0x48 cp hl,_d_
- [ ] 0x4c test hl
- [ ] 0x4d neg hl
- Ldp, llcall, lljp, ldf
- Anything involving the 32 bit index registers ps
- ld xpc,hl, ld hl,xpc
- Anything involving bcde, jkhl, jk

I'm very much feeling the working on ticks pain....

pauloscustodio, 2023-09-22:
List of issues found up to now: (please tick as they are solved)
[ ] (ticks) -mr4k: ```7F AF``` should be ```AND A```, is ```MULU```

pauloscustodio, 2023-09-24:
(ticks) -mr4k: add sp, d (27 81) does not implement carry:

```
                    ld        sp,$1000                      ;[001d] 31 00 10
pc=0020, [pc]=27,    bc=0000,  de=0000,  hl=0000,  af=0044, ix=0000, iy=0000
sp=1000, [sp]=0000, bc'=0000, de'=0000, hl'=0000, af'=0040
f: S=0 Z=1 H=0 P/V=1 N=0 C=0
clockh=CCCC, clockhl=CCCC
                    add       sp,-$7f                       ;[0020] 27 81
pc=0022, [pc]=F5,    bc=0000,  de=0000,  hl=0000,  af=0044, ix=0000, iy=0000
sp=0F81, [sp]=0000, bc'=0000, de'=0000, hl'=0000, af'=0040
f: S=0 Z=1 H=0 P/V=1 N=0 C=0
clockh=CCCC, clockhl=CCCC
```

Adding a negative number should set the Carry flag.

suborb, 2023-09-24:
> (ticks) -mr4k: add sp, d (27 81) does not implement carry:
>
> Adding a negative number should set the Carry flag.

I assumed it would set carry when it passed over the 0x0000/0x10000 boundary since that matches other `add` instructions.

```
                    ld        sp,$000a                      ;[0000] 31 0a 00
pc=0003, [pc]=27,    bc=0000,  de=0000,  hl=0000,  af=0040, ix=0000, iy=0000
sp=000A, [sp]=0000, bc'=0000, de'=0000, hl'=0000, af'=0040
f: S=0 Z=1 H=0 P/V=0 N=0 C=0
clockh=0ABD, clockhl=0070
                    add       sp,-$14                       ;[0003] 27 ec
pc=0005, [pc]=AF,    bc=0000,  de=0000,  hl=0000,  af=0041, ix=0000, iy=0000
sp=FFF6, [sp]=0000, bc'=0000, de'=0000, hl'=0000, af'=0040
f: S=0 Z=1 H=0 P/V=0 N=0 C=1
clockh=0ABD, clockhl=0070
```

```
                    ld        sp,$fff0                      ;[0000] 31 f0 ff
pc=0003, [pc]=27,    bc=0000,  de=0000,  hl=0000,  af=0040, ix=0000, iy=0000
sp=FFF0, [sp]=0000, bc'=0000, de'=0000, hl'=0000, af'=0040
f: S=0 Z=1 H=0 P/V=0 N=0 C=0
clockh=A2D5, clockhl=009A
                    add       sp,+$28                       ;[0003] 27 28
pc=0005, [pc]=AF,    bc=0000,  de=0000,  hl=0000,  af=0041, ix=0000, iy=0000
sp=0018, [sp]=0000, bc'=0000, de'=0000, hl'=0000, af'=0040
f: S=0 Z=1 H=0 P/V=0 N=0 C=1
```

pauloscustodio, 2023-09-24:
Adding a negative number crosses the 0x0000/0x10000 boundary, see below example for standard ```add``` on ```-mz80```:

```
ld hl, $4000
ld de, -$7f
add hl, de
jp 0
```

```
pc=0000, [pc]=21,    bc=0000,  de=0000,  hl=0000,  af=0040, ix=0000, iy=0000
sp=0000, [sp]=0021, bc'=0000, de'=0000, hl'=0000, af'=0040
f: S=0 Z=1 H=0 P/V=0 N=0 C=0
                    ld        hl,$4000                      ;[0000] 21 00 40
pc=0003, [pc]=11,    bc=0000,  de=0000,  hl=4000,  af=0040, ix=0000, iy=0000
sp=0000, [sp]=0021, bc'=0000, de'=0000, hl'=0000, af'=0040
f: S=0 Z=1 H=0 P/V=0 N=0 C=0
clockh=3177, clockhl=5F50
                    ld        de,$ff81                      ;[0003] 11 81 ff
pc=0006, [pc]=19,    bc=0000,  de=FF81,  hl=4000,  af=0040, ix=0000, iy=0000
sp=0000, [sp]=0021, bc'=0000, de'=0000, hl'=0000, af'=0040
f: S=0 Z=1 H=0 P/V=0 N=0 C=0
clockh=3177, clockhl=5F50
                    add       hl,de                         ;[0006] 19
pc=0007, [pc]=C3,    bc=0000,  de=FF81,  hl=3F81,  af=0069, ix=0000, iy=0000
sp=0000, [sp]=0021, bc'=0000, de'=0000, hl'=0000, af'=0040
f: S=0 Z=1 H=0 P/V=0 N=0 C=1
clockh=3177, clockhl=5F50
```

suborb, 2023-09-25:
I see your point, it just ends up not being very useful with it reflecting the sign of the constant second operand?

I took a look at ucsim from sdcc and it has this:

```
cl_rxk::ADD_SP_d(t_mem code)
{
  i8_t d= fetch();
  u32_t res= rSP+d;
  class cl_cell8 &f= destF();
  u8_t forg= f.R();
  cSP.W(res);
  if (res > 0xffff)
    f.W(forg|= flagC);
  else
    f.W(forg&= ~flagC);
  tick(3);
  return resGO;
}
```

pauloscustodio, 2023-09-25:
If the number is sign-extended before the addition (i.e. "if (d & 0x80) d |= 0xFF00;"), then the check of crossing the 0x0000/0x10000 boundary passes and the carry flag is set.

pauloscustodio, 2023-09-25:
I've fixed ticks with: ```v = sp + (((get_memory(pc++) ^ 128) - 128) & 0xffff);```

suborb, 2023-09-25:
It's an ambiguous hole in the docs (and we don't use it), but what exactly does carry mean here? Running through some tests with your code above it seems to be both:

1. The stack pointer decreased
2. The stack pointer wrapped around

Which would make using the flag tricky to detect overflow when establishing a stack frame.

pauloscustodio, 2023-09-25:
> it also appears to be different behaviour to that of ucsim

Having no access to real hardware, I think this is a strong point. But the fact that ```add hl, de```, with DE negative have a different behaviour, is also a strong point.

I can revert the change in ticks and modify the tests to pass on a different condition, if there is no other opinion, or better fact.

suborb, 2023-09-25:
I would err on the same implementation as ucsim. However:

[image]

(from https://www.mouser.com/datasheet/2/111/0190069_p-471326.pdf) even Digi aren't sure how the instruction should work :)

We have even more fun with the gameboy equivalent, from Mame:

```
/*
 *   Z - Reset.
 *   N - Reset.
 *   H - Set or reset according to operation.
 *   C - Set or reset according to operation.
 */

        {
        int32_t n;

        n = (int8_t) mem_read_byte( m_PC++ );

        if ( ( m_SP & 0xFF ) + (uint8_t)(n & 0xFF) > 0xFF )
        {
                m_F = FLAG_C;
        }
        else
        {
                m_F = 0;
        }

        if ( ( m_SP & 0x0F ) + ( n & 0x0F ) > 0x0F )
        {
                m_F |= FLAG_H;
        }

        m_SP = (uint16_t) ( m_SP + n );
        }
```

and ucsim:

```
int cl_gb80::inst_add_sp_d(t_mem code) {
  u16_t  d = fetch( );
  /* sign-extend d from 8-bits to 16-bits */
  d |= (d>>7)*0xFF00;

  regs.raf.F &= ~(BIT_ALL);  /* clear these */
  if ((regs.SP & 0x0FFF) + (d & 0x0FFF) > 0x0FFF)
    regs.raf.F |= BIT_A;
  if (regs.SP + (int)(d) > 0xffff)
    regs.raf.F |= BIT_C;

  regs.SP = (regs.SP + d) & 0xffff;

  tick(15);
  return(resGO);
}
```

suborb, 2023-09-26:
As a heads up, I've got a branch that implements the EDMR internal register (and ioi, ioe in general). 

Page 299 of the user manual (R4000UM.pdf):

[image]

This means that ticks -mr4k will start up in Rabbit 2/3k compatibility mode (as per the real hardware). The +test crt0 for Rabbit4k switches to 4k mode as the first instruction.

suborb, 2023-10-01:
I _think_ I've managed I've either got an implementation or marked as UNIMPLEMENTED all the Rabbit opcodes. I've pulled out (most of) the CPU specific instructions into separate files which I think makes the code slightly easier to reason with.

There will be bugs (I'm sure), but I need to do something else for a while before I head back into there.

pauloscustodio, 2023-10-02:
I think I have finished the z80asm implementation of the R4K/R5K opcodes (except the multi-segment ones LJP, LCALL, LRET, LLCALL, LLRET, LLJP, SJP).

Please let me know of any issue.

@suborb, can the issue be closed, or is there anything missing in ticks or in the library?

suborb, 2023-10-02:
I'll need to build the library for r4k and get ticks running the tests.

I've found a few missing instructions (of course) in ticks which will come through this evening.

If you're done, feel free to merge and I'll pick up using it.

pauloscustodio, 2023-10-02:
I've merged already.

suborb, 2023-10-02:
@pauloscustodio I think these are valid opcodes:

```
        ld      de',bc
        ld      de',de
```

But they're reporting errors with -mr4k but not with -mr2ka

suborb, 2023-10-02:
The library is building (with a couple of BODGEs for above) and tests are running so I'm done for the moment.

pauloscustodio, 2023-10-02:
z80asm fixed.

suborb, 2023-10-03:
Thanks Paulo. I think we're complete here so I'm closing

commit 4f5a026d (2023-10-02): Fix #2388 (z80asm) Rabbit add missing instructions

Add missing instructions on Rabbit:
   ld      de',bc
   ld      de',bc

Add emulation for all other "ld dd',dd" and "ld dd,dd'" on Rabbit.
closed 2023-10-03 by suborb